NaitLee/Cat-Printer #1: I'm working on an upgrade, expecting better stuffs

NaitLee, 2022-03-21:
It's a long time not updated this repo.
Half years ago I got a laptop (now using) but it's not my development platform at that time, while I need bluetooth on it.
So after a rush the code is very poor, in the sight from 2022 😶‍🌫️

Now that I'm cleaning up the main python script while making a brand new frontend -- the earlier is done, now the later
Going to commit them all together, because at the moment there's no actual difference at all.
I'm also planing to host this project (also) to somewhere else. Here really have problems accessing GitHub.

I see [python-for-android](https://python-for-android.readthedocs.io/en/latest/) while investigating bleak (the bluetooth lib), found that I can build an *working* Android version with just what a PC version will have -- Python, basehttp, bleak, and a common web interface.
This way we are all free from the original weird app...
And, *feel free to discuss what you guys really need in an "app"* 😃

Emm, maybe the image monochrome filter needs enhancement as well...

Wish I'll complete this goal 😉

NaitLee, 2022-03-24:
This day I costed 10 hours investigating python-for-android, and bleak integration with it.
It's a long process suffering from the dirty job made by those java developers, but I finally made it...
My test python script successfully being compiled to an apk, installed on my phone, printed the pre-defined data.
There are many, many hacks made for the work. Going to summarize these later.
Well, let's see the final result...

frankenstein91, 2022-03-26:
@NaitLee in an other project we see CUPS as a good option
https://github.com/rbaron/catprinter/issues/24

NaitLee, 2022-03-27:
**Hey, current version of this project has a very basic CUPS/IPP support** 😄
You can see some related code at [server.py](https://github.com/NaitLee/Cat-Printer/blob/bde24855e09230340bcd6185649d9c25cbd4e61a/server.py#L79). I just kept it secret for a long time...
Though, it's *really very* basic, and it can only receive printed content and print it, no further "status" at all.
It requires ghostscript being installed, and add the `ipp://127.0.0.1:8095/GB02` as a PostScript printer. (replace GB02 as your printer bluetooth name)
It directly invokes `gs` in the current shell. Be sure the `gs` is not `alias`-ed by some advanced shells to `git status`... (if it happens, try `unalias gs`)
Now just launch `python3 server.py`, enable bluetooth and open the printer.
Try it with something like LibreOffice! Use an A4 portrait paper, and a bigger font (probably more than 32pt)

CUPS/IPP is kinda complex. If it's going to be fully supported, around 80% of the code will be of that.
You can see the IPP specification [here](https://datatracker.ietf.org/doc/html/rfc8010). It's not something short after all.

Worthy to mention is that this project is different: it aims at a user interface, not just backend. So the desired user-friendliness is already on the Web interface, and thanks to Web you can actually copy-paste almost anything to it, then print.
Going to make it better after all the main jobs, at least mimicking a stable just-work CUPS interface, if ya reilly want 😜

NaitLee, 2022-03-27:
I'm wondering if I should make all the stuffs (like "full" wysiwyg editing and stickers/borders etc) or just keep it simple.
There are all editors around (libreoffice, gimp, inkscape, etc), while most of them are on PC. Some Android users will still feel "uncomfortable", though.
I tried fabric.js and html2canvas previously, and I think making full use of them here have some chance of wasting the time.
"A software should only do one thing, and do it well." But users won't take it serious. What a dilemma.
I know I can't satisfy everyone, and that's a reason for me to make web interface, satisfying low-end poor users.
I'll just do the basic now... Going to deal with these later...

frankenstein91, 2022-03-27:
I would love to see templates for barcodes, business cards, shopping lists and route information.

NaitLee, 2022-03-28:
You mean a full interface 😃
Well, after some thoughts I think I should switch the direction: just make a full featured Printer *frontend*, and let *backends* can integrate with it. And there are already many versions of backends across here, just let them expose some functions or a class, or I adapt them.

I'd tell the full reason later. But you need to understand the situation, of the dirty "[intellectual property](https://www.gnu.org/philosophy/not-ipr.html)".
In doubt? Read the EULA of iPrint app. (if have ability, read EULA of 精准学习 the Chinese/original version.) I don't need to tell further if you [spot the points](https://www.gnu.org/philosophy/essays-and-articles.html#Laws). *I did nothing wrong*, but it's still too big risk for me a native person, and local users.
Let's help [Free Software](https://www.gnu.org/philosophy/free-software-even-more-important.html), while keeping everyone well...

brysonreece, 2022-03-29:
If it's possible, the addition of `GB03` support would be nice - I haven't found any project compatible with that model yet, which is what I unknowingly ordered. For reference, [here's a link to Amazon](https://www.amazon.com/dp/B097XX9F7D) where the `GB03` model can be purchased.

Another differentiating factor I've noticed is that the `iPrint` app has to be launched and printer chosen before the printer will show up in my phone's Bluetooth Devices menu. Despite all efforts, my Mac refuses to acknowledge the printer is broadcasting at all and refuses to list the device, even with an external BLE dongle.

NaitLee, 2022-03-30:
From [Free Software Is Even More Important Now](https://www.gnu.org/philosophy/free-software-even-more-important.html):

>  ... It also wrongs others if you make a promise not to share. It is evil to keep such a promise, and a lesser evil to break it; to be truly upright, you should not make the promise at all.

Fuck it up (╯⎺Д⎺)╯︵ ┻━┻
Let's do it.
```
$ jadx iprint.apk
```

I've found that in GB03 model the Bluetooth command buffer sent to printer is "compressed" with a small algorithm.
Along with this there are some other things that different.
It is really nothing to implement these shit, but we need a physical printer to check if it works.
Amazon did a bad job for this, it's 2x-4x money to buy one. Completely NOT affordable for me.
I've consulted local seller where I bought my GB02, they say they have no GB03.
Compared to other, these printers are *very* cheap -- I bought one for around ￥80 (\~$12.5), and now they've discounted to ￥40 (\~$6)
They have more than cat printers. In fact they have everything, and some printers are even AI compatible.
That's a lot of work, if going to do all of them.

Also, to check what's up in BLE we need some special tools.
I've found [this](https://infocenter.nordicsemi.com/topic/ug_sniffer_ble/UG/sniffer_ble/intro.html). I bought a compatible dongle just now for ￥60 (\~$10) but I suddenly think I have no idea to make use of it.

Sorry for the inconvenience, the biggest obstacle is money, and communication.
I want anonymity for continuing further jobs, and of course users' collaboration of sharing/testing the printers...
Any brave guy to provide such a solution?
Totally unbelievable such a job turned to be on my head... 😶‍🌫️

... By the way the new frontend is going to be usable. Now it have better monochrome algorithm.
Doing the finalization works... I think putting it on a new branch could be fine... 😇

frankenstein91, 2022-03-30:
I ordered a printer yesterday on AliExpress... I don't know which version it would be... But I think I will be brave enough for testing 🙂

claudioruiz, 2022-03-30:
> 

Oh crap, maybe that's the reason this is not working to me. I just bought a model and it seems to be GB03. How different is it from the older models?

NaitLee, 2022-03-31:
I'm wondering if I could:
1. Get/Fake a programmable BLE device
2. Program the device, lead iPrint to regard it as a certain working model (in our case GB03)
3. Do actions with iPrint (like print), but our device actually dump the traffic
4. Analyze the traffic, try to make a equivalent in our way

Well, the [nRF52840](https://www.nordicsemi.com/Products/nRF52840) dongle I've bought have use now... 😜

In addition there's also a "canPrintLabel" stuff that happens in GB03's bitmap procedure.
Along with compression and detection, it's very different...

In their code those Chinglish function names really suck. Give them some color to see see.

NaitLee, 2022-04-06:
Sorry for the long wait, but it's finally released.
You can test the 0.1.0 releases immediately, and play with the Android version!

At the moment there's few function. Going to do them later.
The Android building how-to is not there yet. Patience...
As for GB03, I'll take a look when I feel appropriate, with the bluetooth dongle hack.
My C programming skill is not good enough, though... (doing with the dongle needs to write C/C++ firmware programs)

claudioruiz, 2022-04-06:
Oh still waiting for that GB03 update! :(

NaitLee, 2022-04-06:
Well, just now I set up the bluetooth low energy development environment, modified an example code (a typical "blink" program on all IoT stuffs, but have bluetooth interface, I changed the broadcast name to "GB03") and successfully fooled iprint to list my dongle 😜
But I need real data transferring techniques, serial communication to dump the traffic, and some analytic tricks, will manage to do/learn all of them.
Every day is a new progress 😄

BTW do you really hate "iprint"? I see it just-so-so. Yes it's big.
My reason to do all of these is I hate the chinese version. It even wants my phone number. And everyone else's.
Of course I fight for true freedom of worldwide 😉

claudioruiz, 2022-04-06:
> BTW do you really hate "iprint"? 

Oh yes I do. The whole point for me of getting this little printer is to be able to manage it on CLI and not install any fishy app on my phone. Thanks @NaitLee for all your work!

NaitLee, 2022-04-07:
For CLI what you need is (currently) just `printer.py`, and the Bleak dependencie.
I also concern about these, while I want to follow the Unix philosophy: do one thing and do it well.
That means it will not bother geeks to force many dependencies. But I also want normal users to enjoy it.

The script accepts a PBM image file, either with path to file or use stdin.
For other formats there's imagemagick, usually already there in GNU/Linux, simply do image manipulation with that (the `convert` command) then pipe to `printer.py`
I once did that but lost the command parameters. It takes some time to get the correct parameters. Going to build it in later.
In the mean while there's actually a procedure for PBM generation in the Web frontend. You can modify the code of either side slightly to dump the result.
There is also GIMP for general image manipulation with GUI. It can export PBM as well.

I come up the idea to support "text printing", you enter the command line, then just type some text (or pipe to stdin!), and printer will print, in a "streaming" way.
For the same reason I want to use the [PF2](http://grub.gibibit.com/New_font_format) font format, it's easy to make (use `grub-mkfont` with a common ttf/otf font file), easy to support, Unicode compatible, and for it's bitmap font it can do the "typewriter" hack!

Also the current `printer.py` is actually just a purged version of legacy version. There are many things to be changed.

Seems now I should put GB03 support on the top. Stay tuned!

claudioruiz, 2022-04-07:
I’m absolutely tuned. My GB03 printer is still waiting to be used. Thanks again!

sickplanet, 2022-04-07:
Hello guys, just got GB03 to print with latest version 0.1.0. easy as f### !

NaitLee, 2022-04-08:
I would like to say something now...

Yesterday I put the effort totally to the dongle hack, but I faced big problems:
- The dongle BLE MTU is statically the default 23. I didn't manage to change it with a 3rd party easy-to-use lib. (Cat Printer have at least 200)
- The dongle can't emulate a USB serial port by default. Thus dumping or debugging is hard in the current sight. I should use my another board as "middleware" of data transferring, but it needs another effort.
- The proprietary SDKs of the dongle really suck. sucks times harder than Android/Java. Just can't believe they give the community an Electron APP for flashing firmware that needs root permission to work (at least on Arch Linux based distros, even given `uucp` group added) and *will segfault at any time*. No pin label to GPIO image/figure at all and I should brute test. And the JLink tool that says missing **DLL** *on Linux*. Last the document/manual all about the dirty license but nothing about how-to for the SDK code. Approx. 1/8 of all the 200MiB SDK C code is comment of their license. Nevermind.

Of course I wanted to by another **open** ble thing that just works. But an idiot brought covid from shanghai to my county and everything messed up, express/delivery stopped and I can't buy any hardware or snacks now.

Too many bad things happened these days. What I can do is just hope everyone well 😢

Today I have other work. If I want I would continue to try on a Windows VM. That's what those suckers love.

NaitLee, 2022-04-08:
Okay, I just found Android itself is capable of BLE emulation. There's nRF Connect APP, going to use it on my another phone now rather than wasting time 🙃

NaitLee, 2022-04-08:
Hi guys, I managed to dump all the data between iprint and GB03 (and "original" GB02 for comparison), and analyze... but found maybe it's just a matter of a `0x12` byte.
*Please test with the attached script below*, by replacing the same file in a working 0.1.0 release with it.
I don't know if it will work yet. Please say...

--------

Be verbosely, though I finally understood how it works, the GB03 "compress" procedure seems being optional, and actually have some chance to make data chunk bigger -- then iprint will rather just send that chunk in the old way, just like e.g. GB02.

But implementing it will be cool and save connection time... "Low Energy", right? 😄

**Mathematics alert!**
*These are found with my eyes on a calculator (KCalc) with bit-mode. The original code is too messy...*
(Semi-)Detailed statement of that compression:
Note: The "compressed" data is in a command `0xBF` rather than normal `0xA2`.
- While parsing bitmap, Pick next chunk of continuous bits, that have the same value (i.e. all are `0` or all are `1`), to a maximum count of 127
- If the chunk size is bigger than 16, do the following:
  - Take a number from 96, 64, 48, 32, 24, 16 (they can all modulus 384 to 0). Adapt the maximum but in the range of original size, except when size is above 96 then just use 96.
  - Divide 384 with the number, get another integer number. Put its value as the command parameter.
  - Allocate a byte, do this:
    - Make the **first bit** `0`, if the chunk is full of `0`. Otherwise `1`.
    - Let the other 7 bits be the size of this chunk. Of course maximum value is 127.
  - Do the same to the other bits... Until the payload is enough big. Put the result in the command payload.
- If the chunk is too small, simply use the old format, that just regard a bit as a pixel.
- Loop 'till end...

Okay, there *maybe* few mistakes... Let's check later...

Examples:
- HEX `FF FF 00 00`: this is the old "uncompressed" format, that represents 16 black pixels and following 16 white pixels.
  - Binary as `1111 1111  1111 1111  0000 0000  0000 0000`
- HEX `C0 40`: this is the "compressed" format, that represents 64 black pixels and following 64 white pixels.
  - Binary as `1100 0000  0100 0000`. It's (somewhat) 8 times smaller! Taste the feeling 😙
  - Especially useful with text printing... There are many continuous white pixels there

--------

Oh right, let me put the attachment now... Decompress it first...
[printer.zip](https://github.com/NaitLee/Cat-Printer/files/8454013/printer.zip)
.

sickplanet, 2022-04-08:
> Hi guys, I managed to dump all the data between iprint and GB03 (and "original" GB02 for comparison), and analyze... but found maybe it's just a matter of a `0x12` byte. _Please test with the attached script below_, by replacing the same file in a working 0.1.0 release with it. I don't know if it will work yet. Please say...
> 
> EDIT: if it doesn't work, please also test if BLE Characteristic is a problem. In the code find & replace this:
> 
> ```
> 0000ae01-0000-1000-8000-00805f9b34fb
> ```
> 
> with one of these. It's better to try all... especially the `ae81` one:
> 
> ```
> 0000ae81-0000-1000-8000-00805f9b34fb
> 0000ab01-0000-1000-8000-00805f9b34fb
> 0000ff02-0000-1000-8000-00805f9b34fb
> 49535343-8841-43f4-a8d4-ecbe34729bb3
> ```
> 
> Be verbosely, though I finally understood how it works, the GB03 "compress" procedure seems being optional, and actually have some chance to make data chunk bigger -- then iprint will rather just send that chunk in the old way, just like e.g. GB02.
> 
> But implementing it will be cool and save connection time... "Low Energy", right? 😄
> 
> **Mathematics alert!** _These are found with my eyes on a calculator (KCalc) with bit-mode. The original code is too messy..._ (Semi-)Detailed statement of that compression: Note: The "compressed" data is in a command `0xBF` rather than normal `0xA2`.
> 
>     * While parsing bitmap, Pick next chunk of continuous bits, that have the same value (i.e. all are `0` or all are `1`), to a maximum count of 127
> 
>     * If the chunk size is bigger than 16, do the following:
>       
>       * Take a number from 96, 64, 48, 32, 24, 16 (they can all modulus 384 to 0). Adapt the maximum but in the range of original size, except when size is above 96 then just use 96.
>       * Divide 384 with the number, get another integer number. Put its value as the command parameter.
>       * Allocate a byte, do this:
>         
>         * Make the **first bit** `0`, if the chunk is full of `0`. Otherwise `1`.
>         * Let the other 7 bits be the size of this chunk. Of course maximum value is 127.
>       * Do the same to the other bits... Until the payload is enough big. Put the result in the command payload.
> 
>     * If the chunk is too small, simply use the old format, that just regard a bit as a pixel.
> 
>     * Loop 'till end...
> 
> 
> Okay, there _maybe_ few mistakes... Let's check later...
> 
> Examples:
> 
>     * HEX `FF FF 00 00`: this is the old "uncompressed" format, that represents 16 black pixels and following 16 white pixels.
>       
>       * Binary as `1111 1111  1111 1111  0000 0000  0000 0000`
> 
>     * HEX `C0 40`: this is the "compressed" format, that represents 64 black pixels and following 64 white pixels.
>       
>       * Binary as `1100 0000  0100 0000`. It's (somewhat) 8 times smaller! Taste the feeling 😙
>       * Especially useful with text printing... There are many continuous white pixels there
> 
> 
> Oh right, let me put the attachment now... Decompress it first... [printer.zip](https://github.com/NaitLee/Cat-Printer/files/8454013/printer.zip) .

I tested and is working... with some random disconnects but it's fine!

frankenstein91, 2022-04-12:
Hi @NaitLee 
I have got my printer today... but I can not get the server.py working
all I see is 
[image]

I can print via printer.py

any idea?

frankenstein91, 2022-04-12:
> **Hey, current version of this project has a very basic CUPS/IPP support** smile You can see some related code at [server.py](https://github.com/NaitLee/Cat-Printer/blob/bde24855e09230340bcd6185649d9c25cbd4e61a/server.py#L79). I just kept it secret for a long time... Though, it's _really very_ basic, and it can only receive printed content and print it, no further "status" at all. EDIT: The newest commit is not released yet. To test the following please `git clone` the repo and get the dependencies (can just pick most from a release) It requires ghostscript being installed, and add the `ipp://127.0.0.1:8095/GB02` as a PostScript printer. (replace GB02 as your printer bluetooth name) It directly invokes `gs` in the current shell. Be sure the `gs` is not `alias`-ed by some advanced shells to `git status`... (if it happens, try `unalias gs`) Now just launch `python3 server.py`, enable bluetooth and open the printer. Try it with something like LibreOffice! Use an A4 portrait paper, and a bigger font (probably more than 32pt)
> 
> CUPS/IPP is kinda complex. If it's going to be fully supported, around 80% of the code will be of that. You can see the IPP specification [here](https://datatracker.ietf.org/doc/html/rfc8010). It's not something short after all.
> 
> Worthy to mention is that this project is different: it aims at a user interface, not just backend. So the desired user-friendliness is already on the Web interface, and thanks to Web you can actually copy-paste almost anything to it, then print. Going to make it better after all the main jobs, at least mimicking a stable just-work CUPS interface, if ya reilly want stuck_out_tongue_winking_eye

I did not get this working as well

NaitLee, 2022-04-12:
Sorry for the inconvenience, the CUPS/IPP feature is in legacy, but not this new version yet. I updated the comment to make it clear.

As for frontend, check if that:
a) you have an old browser, that doesn't support newer ecma script features, like firefox esr. *and*
b) you are using the repo alone, without putting i18n script and/or transpiling a compatibility script (see the points in development.md), *or*
c) you didn't enable javascript

anyway, press F12 in browser and see what's inside the console.
you may try with a new browser and a published release (like 0.1.0)

frankenstein91, 2022-04-12:
d) VPN somehow

NaitLee, 2022-04-12:
> It didn't work either. Still waiting for the support! Thanks again

Oh, no 😢
Maybe I should check if iprint sent something during initialization...
Just tough.

frankenstein91, 2022-04-12:
@claudioruiz did I get it... you have an none working GB03?
I have a GB03 too... works great

claudiorrrr, 2022-04-12:
> @claudioruiz did I get it... you have an none working GB03? I have a GB03 too... works great

Yup, no luck at all even connecting the printer to the computer. It seems as the printer doesn't even exist for the computer (I did test it with my phone using nRF Connect and it's capable to find it). I can't use it so far :(

frankenstein91, 2022-04-12:
@claudiorrrr could you give me the output of `lsusb | grep Bluetooth`

claudiorrrr, 2022-04-12:
> @claudiorrrr could you give me the output of `lsusb | grep Bluetooth`

`lsusb | grep Bluetooth` shows no output on MacOS. Same with Ubuntu on a x220. :/

NaitLee, 2022-04-12:
i remember bleak said macos is different in some cases. you can look into it in its document site...
and the printer actually isn't a 'normal' bluetooth stuff. 'normal' connection doesn't work on it, and pairing does nothing useful.
Bluetooth Low Energy should not be confused with 'normal' bluetooth though... you can know more on internet...

frankenstein91, 2022-04-12:
> > @claudiorrrr could you give me the output of `lsusb | grep Bluetooth`
> 
> `lsusb | grep Bluetooth` shows no output on MacOS. Same with Ubuntu on a x220. :/

Try to change usb to pci

claudiorrrr, 2022-04-12:
> > > @claudiorrrr could you give me the output of `lsusb | grep Bluetooth`
> > 
> > 
> > `lsusb | grep Bluetooth` shows no output on MacOS. Same with Ubuntu on a x220. :/
> 
> Try to change usb to pci

same

NaitLee, 2022-04-13:
I'm wondering if it's just problem of Mac.
If could, you can try with a Linux VM in Mac. I recommend [Virtualbox](https://www.virtualbox.org/).
It's easy to do basics. And I just mention that you need to *attach the bluetooth adapter to VM* in virtualbox (in my case it's called "Intel Corp." in USB devices selection), and to do this you likely need "virtualbox extension pack" installed first.
In Linux env you can follow the readme to get work. Don't forget check the common myths like didn't enable bluetooth etc.
The printer is able to be scanned & shown, but not able to be connected nor paired (with 'common' bluetooth protocol). usb cable does no work.
I will publish a pre-release now, including current code, for testing...
Still, feel free to ask any problems encountered 😄

frankenstein91, 2022-04-13:
I had the idea that the Bluetooth adapter isn't BLE capable... But it wasn't there as a USB or PCI device. So we can't really check that.

NaitLee, 2022-04-13:
> I had the idea that the Bluetooth adapter isn't BLE capable... But it wasn't there as a USB or PCI device. So we can't really check that.

Well, why not just test 🙃 My bluetooth adapter is built inside my computer, but still shown as USB device.
Apple MacOS alike is "painful", said by many (at least) web developers, for it's *lockdown* nature. But Mac computer is yet expensive. I never wanted to waste money on such a stuff... 😅

And BLE have some concepts, thing like our printer is called "peripheral", and our computer is called "central" when it controls the printer.
My computer bluetooth adapter ~~is not BLE peripheral compatible either (unsure), but it~~ can communicate to printer! I can even see the traffic in Wireshark.
My cheap bluetooth earbuds are identified as BLE too. If there is AirPods, why isn't Mac computer BLE compatible 🤫
Android phones can mostly do *both*, though just one at once. That's how I've dumped iprint traffic with 2 phones 😊

I will also put apk in the release, for also trying on android.
Also in the pre-release there's cups/ipp. Try that, please 😉