NaitLee/Cat-Printer #21: RuntimeError: no running event loop

claudiorrrr, 2022-05-02:
Hi, with the new version this is not working anymore. When I run server.py and go to the browser, the app shows an error message. This is the output at CLI

I don't know how to sort it out (on MacOS)

```
c@m1 ~/g/Cat-Printer (main)> python3 server.py
Serving at http://127.0.0.1:8095/
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 63122)
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/Users/c/github/Cat-Printer/server.py", line 276, in finish_request
    self.handler.__init__(request, client_address, self)
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/socketserver.py", line 747, in __init__
    self.handle()
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/server.py", line 425, in handle
    self.handle_one_request()
  File "/Users/c/github/Cat-Printer/server.py", line 82, in handle_one_request
    super().handle_one_request()
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/server.py", line 413, in handle_one_request
    method()
  File "/Users/c/github/Cat-Printer/server.py", line 229, in do_POST
    self.handle_api()
  File "/Users/c/github/Cat-Printer/server.py", line 190, in handle_api
    } for device in self.printer.scan()]
  File "/Users/c/github/Cat-Printer/printer.py", line 367, in scan
    scanner = BleakScanner()
  File "/Users/c/github/Cat-Printer/venv/lib/python3.9/site-packages/bleak/backends/corebluetooth/scanner.py", line 44, in __init__
    self._manager = CentralManagerDelegate.alloc().init()
  File "/Users/c/github/Cat-Printer/venv/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 61, in init
    self.event_loop = asyncio.get_running_loop()
RuntimeError: no running event loop
----------------------------------------
Exception ignored in: <CentralManagerDelegate objective-c instance 0x0>
Traceback (most recent call last):
  File "/Users/c/github/Cat-Printer/venv/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 102, in __del__
    self.central_manager.removeObserver_forKeyPath_(self, "isScanning")
AttributeError: 'CentralManagerDelegate' object has no attribute 'central_manager'
Exception ignored in: <function CentralManagerDelegate.__del__ at 0x10418ce50>
```

closed 2022-05-02 by claudiorrrr

claudiorrrr, 2022-05-02:
Ha, I closed bc it started working after I reboot :/

NaitLee, 2022-05-02:
Oh, okay 😄