NaitLee/Cat-Printer #23: Add support for YT01 model

williamd1k0, 2022-05-04:
I bought this model recently and ended up getting a little frustrated because the only way to use it was through a suspicious mobile app.
I will be very grateful if you can add support for this model. I can help you with any information, testing and maybe even coding.

Recommended app: (iBleem) https://play.google.com/store/apps/details?id=com.bleem.liugm

Thank you so much for working on this project.

NaitLee, 2022-05-04:
I see it's yet another brand.
I'm gonna try [another](https://apkpure.com/fun-print/com.fun.mxw) app, because it's a) by same vendor, b) looks cleaner, c) seems made for global
You could also just try it, in the mean time while when I investigate this model.
(of course, no need if you really don't want)

off-topic: the vendor name seems means chinese pinyin "印题宝". though I don't want know further about it.

NaitLee, 2022-05-04:
Hi, you may help me to find the Bluetooth "Characteristics" of YT01.
Use [nRF Connect](https://apkpure.com/nrf-connect-for-mobile/no.nordicsemi.android.mcp), open the printer, find it with "Scanner" and see what's inside.

(the original app has really bad UI logic)

williamd1k0, 2022-05-04:
>(the original app has really bad UI logic)

Yeah :sweat_smile:, I've tried this "Fun Print" now and it's actually the same as iBleem with slight differences in the UI.

[image]

Some pics:
[image]
[image]

>Use [nRF Connect](https://apkpure.com/nrf-connect-for-mobile/no.nordicsemi.android.mcp), open the printer, find it with "Scanner" and see what's inside.

Ok, I will try it out.

williamd1k0, 2022-05-04:
<details>
<summary><b>Screenshots</b></summary>

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

</details>

<details>
<summary><b>Log 2022-05-04 02_47_13.txt</b></summary>

```
nRF Connect, 2022-05-04
YT01 (33:9F:3E:E3:1F:63)
V	02:45:45.750	Connecting to 33:9F:3E:E3:1F:63...
D	02:45:45.750	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D	02:45:46.935	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	02:45:46.974	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	02:45:46.975	Connected to 33:9F:3E:E3:1F:63
D	02:45:46.980	wait(1600ms)
V	02:45:48.582	Discovering services...
D	02:45:48.582	gatt.discoverServices()
D	02:45:48.617	[Callback] Services discovered with status: 0
I	02:45:48.617	Services discovered
V	02:45:48.669	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
Unknown Service (0000ae30-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [WNR] (0000ae01-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000ae02-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [WNR] (0000ae03-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000ae04-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [I] (0000ae05-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (0000ae10-0000-1000-8000-00805f9b34fb)
Unknown Service (0000ae3a-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [WNR] (0000ae3b-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000ae3c-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
D	02:45:48.670	gatt.setCharacteristicNotification(0000ae02-0000-1000-8000-00805f9b34fb, true)
D	02:45:48.673	gatt.setCharacteristicNotification(0000ae04-0000-1000-8000-00805f9b34fb, true)
D	02:45:48.677	gatt.setCharacteristicNotification(0000ae05-0000-1000-8000-00805f9b34fb, true)
D	02:45:48.681	gatt.setCharacteristicNotification(0000ae3c-0000-1000-8000-00805f9b34fb, true)
```

</details>

NaitLee, 2022-05-04:
Oh thank you, it's very detailed 😄
Next step for me is to trick the app to send data, dump it & analyze.
The tool is also nRF Connect.
And I feel these steps are universal and worth to make a tutorial later 🙃

The printer is completely different, maybe including communication "magic data".
Let me gather more...

NaitLee, 2022-05-04:
Oh, wait, Services/Characteristics UUID needs to be full.
Long tap on each entry (and sub-entry) in connected YT01 "Client", on right-top there's copy button.
Put them in a text file, and be sure to label corresponding Properties (read, write, notify, etc)
It's a repeating process, but maybe worthwhile 😉

NaitLee, 2022-05-04:
Some side notes:

- the original app is actually in JavaScript
- it also support many other models, of their own
- the real code is in `assets` (of apk). but obfuscated
- rough stat: more than 50 js files, each is around 100kib in average. after prettify there will be 10,000 lines per file
- that doesn't mean it's better than similar things: bad naming convention, pinyin variable names & english typos, poorly managed processing logic (aka Spaghetti Code, for example using `if` everywhere for different printer models)
- above is got with minified code. just think about how's their codebase.
- the communication "magic data" is buried in these obscure code. it will take some time to dig its logic out, but our traffic dumping technique may improve efficiency
- inside, the only thing I feel good about is i18n

In order to really support all `YTXX` I think we need more people, for sharing & testing.
But I have no ability to do them all by myself 😢

I will continue slowly.

williamd1k0, 2022-05-05:
>It's a repeating process, but maybe worthwhile 

Thanks to KDE Connect I was able to share the text directly to my PC instead of copying/pasting everything manually :sweat_smile: 

Let me know if anything is still missing.

```yaml
Generic Access:
  UUID: 0x1800
  PRIMARY SERVICE:
    Device Name:
      UUID: 0x2A00
      Properties: READ, WRITE
      Parsed value: YT01
      Raw value: (0x) 59-54-30-31, "YT01"

Unknown Service:
  UUID: 0000ae30-0000-1000-8000-00805f9b34fb
  PRIMARY SERVICE:
    Unknown Service:
      UUID: 0000ae01-0000-1000-8000-00805f9b34fb
      Properties: WRITE NO RESPONSE
    Unknown Service:
      UUID: 0000ae02-0000-1000-8000-00805f9b34fb
      Properties: NOTIFY
    Unknown Service:
      UUID: 0000ae03-0000-1000-8000-00805f9b34fb
      Properties: WRITE NO RESPONSE
    Unknown Service:
      UUID: 0000ae04-0000-1000-8000-00805f9b34fb
      Properties: NOTIFY
    Unknown Service:
      UUID: 0000ae05-0000-1000-8000-00805f9b34fb
      Properties: INDICATE
    Unknown Service:
      UUID: 0000ae10-0000-1000-8000-00805f9b34fb
      Properties: READ, WRITE
      Parsed value: (0x) 00-00-00-00
      Raw value: (0x) 00-00-00-00

Unknown Service:
  UUID: 0000ae3a-0000-1000-8000-00805f9b34fb
  PRIMARY SERVICE:
    Unknown Service:
      UUID: 0000ae3b-0000-1000-8000-00805f9b34fb
      Properties: WRITE NO RESPONSE
    Unknown Service:
      UUID: 0000ae3c-0000-1000-8000-00805f9b34fb
      Properties: NOTIFY
```

NaitLee, 2022-05-05:
I see they are just *very* alike.
Please test if this brute way would work:
- Pick a release, or clone repo & install deps
- Open file `printer_lib/models.py`, inside add a model `YT01` as the same as *any* other:
   ```python
    Models = {
        'YT01': Model(384, False),
        'GB01': Model(384, False),
        # ...
    }
   ```
- Try to print...

NaitLee, 2022-05-05:
Ah great 😄
So next times I should not overlook other models. They have big chance to be the same 😅
But who knows when I will meet a really different one...

williamd1k0, 2022-05-05:
>Off-topic: our laptop is very similar

Hehe, mine is a Lenovo Ideapad S145.

So, I will close the issue now. Thanks again for working on this project, it's really helpful.
I will definitely help improve the app (UI, features, i18n etc) in the future, one thing I was looking forward to before buying the printer was this specific project: https://github.com/receiptline/receiptline (I really like to automate things through the command-line :sweat_smile:).

closed 2022-05-05 by williamd1k0

braintapper, 2022-05-31:
FWIW, I was able to get my cat printer model (MX05) to work using the YT01 settings.

NaitLee, 2022-05-31:
So there are many "unknown but supported" models.
That may mean I should put a "Test other devices" feature, with existing settings :)