NaitLee/Cat-Printer #28: set_energy is likely wrong

X-Ryl669, 2022-05-06:
If I understand your code correctly, the set_energy function is using the same action byte as set_mode function (0xBE). From the other code around, it seems it should be 0xAF instead. 

Yet, this project gives the best output so far compared to other repo, so I don't know who's right or wrong here.

NaitLee, 2022-05-06:
Oh, you are right 😄
By changing to `0xaf`, "energy" worked on my printer.
Maybe I've mistaken something at earlier times.

I'd make it as an option later. Thank you!

NaitLee, 2022-05-07:
I want to thank you again 😁
This idea made me investigate further about those mysterious printer commands,
and I've found something great!

The newest 0.5.0 release won't disappoint you!

NaitLee, 2022-05-25:
This concern is done in 0.5, called "strength", and adjustable :)

closed 2022-05-25 by NaitLee

X-Ryl669, 2022-05-25:
I'm sorry for not answering earlier. The printer doesn't work at home, it fails with Bleak errors on both a linux system and a mac os system, so I gave up and used the Android's app here. It works on my work's computer (linux), so it's not the printer or the application code fault. I'm not sure I should report bleak errors on this repository anyway.

X-Ryl669, 2022-06-04:
Here's what it gives:
```
callApi/<@http://192.168.0.132:8095/~every.js:704:28

{
    "name": "Exception",
    "details": ""
}
callApi/<@http://192.168.0.132:8095/~every.js:704:28

{
    "name": "org.bluez.Error.Failed",
    "details": "Software caused connection abort"
}
callApi/<@http://192.168.0.132:8095/~every.js:704:28

{
    "name": "org.bluez.Error.Failed",
    "details": "Software caused connection abort"
}
callApi/<@http://192.168.0.132:8095/~every.js:704:28

{
    "name": "org.bluez.Error.Failed",
    "details": "Software caused connection abort"
}
```
On the console, I have:
```
Exception happened during processing of request from ('192.168.0.84', 50616)
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "./server.py", line 315, in finish_request
    self.handler.__init__(request, client_address, self)
  File "/usr/lib/python3.8/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.8/http/server.py", line 427, in handle
    self.handle_one_request()
  File "./server.py", line 95, in handle_one_request
    super().handle_one_request()
  File "/usr/lib/python3.8/http/server.py", line 415, in handle_one_request
    method()
  File "./server.py", line 258, in do_POST
    self.handle_api()
  File "./server.py", line 233, in handle_api
    self.printer.connect(name, address)
  File "/home/cyril/Cat-Printer/printer.py", line 343, in connect
    self.loop(
  File "/home/cyril/Cat-Printer/printer.py", line 314, in loop
    results.append(self._loop.run_until_complete(future))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/cyril/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 267, in connect
    reply = await asyncio.wait_for(
  File "/usr/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```
Here's the output of the printer.py command:
```
$ python3.8 ./printer.py -s 5 -m -c text flyingBirdBig.png
Cat Printer
Connecting
Traceback (most recent call last):
  File "./printer.py", line 745, in <module>
    main()
  File "./printer.py", line 725, in main
    _main()
  File "./printer.py", line 706, in _main
    printer.scan(identifier, use_result=True)
  File "./printer.py", line 380, in scan
    self.connect(devices[0].name, devices[0].address)
  File "./printer.py", line 343, in connect
    self.loop(
  File "./printer.py", line 314, in loop
    results.append(self._loop.run_until_complete(future))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/cyril/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 120, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address 6D:88:80:AE:CB:BE was not found.
sys:1: RuntimeWarning: coroutine 'BleakClientBlueZDBus.start_notify' was never awaited
```

It finds the printer but fails to connect with it.

NaitLee, 2022-06-04:
As you've said maybe it's not problem of printer nor app.

Try to use other kind of devices, for example Bluetooth ear buds, see if that's possible 🙂

If not then adapter is bad. (I'm not sure if all adapters can do Bluetooth Low Energy, anyway)