NaitLee/Cat-Printer #33: Publish the Android app to F-Droid

doragasu, 2022-06-03:
Thanks for this application, I got my printer yesterday, tested the app and it works wonders!

Could you please publish it to F-Droid repository? (In case you do not know it, it is a repository for free and open source apps only).

The instructions can be found in the "Submit Applications" section: https://f-droid.org/en/contribute/

Thanks!

NaitLee, 2022-06-03:
Thanks for the concern.

Publishing on F-Droid is a wish I have from very start :)

For now the biggest obstacle is an automated build config, required by F-Droid.
I think I have no ability. The complexity sucks.
I want help on this kind of matters. I'd write a guide about preparing build environment in the future. (It's totally manual)

Days ago I discovered an (unrelated but I like) project/repo: [obfusk/jiten](https://github.com/obfusk/jiten)
Comparing to Cat-Printer, this have almost the same way to Android: `python-for-android` & its webview bootstrap
-- *but it's on F-Droid!* Thus worth to have a look on/ask for how it's done.

Other reasons include maybe incomplete features etc. Things are still on-going.

doragasu, 2022-06-06:
Hi!

Thanks for your response, it's nice reading you already are interested in publishing to F-Droid!

I wish I could help with this, unfortunately I have no experience with Android development. I hope you can find a way to solve the problem, or someone with the skills can step in to help.

Again, thanks for the app!

doragasu

NaitLee, 2023-02-12:
There’s a third-party F-Droid repo that included Cat-Printer: https://android.izzysoft.de/repo

From its [info](https://android.izzysoft.de/repo/info#updates), Cat-Printer (along with other apps) will keep updated there by automation.

Just discovered that by diving into the author of #48 😄

doragasu, 2023-02-12:
Thanks for sharing, that can be a solution for many users. For me, I'm a bit reticent to add that repository, since it contains a lot of packages, many of them already included in the official F-Droid repo.