NaitLee/Cat-Printer #37: eInk display

frankenstein91, 2022-09-03:
I tested the app today on an Android tablet with an eInk display. Unfortunately there seems to be no difference between dark and light design. On the tablet, light would be much better.

NaitLee, 2022-09-03:
I'm wondering if:

- The system have Dark theme enabled, system-wide & affecting the embedded WebView, then Cat-Printer interface is always dark because of CSS `@media` adaption (My android phone is of such. Cat-Printer have no forcible “Bright Theme” at the moment)
- The e-paper have low or no black/white resolution, so Cat-Printer default background (`#eee`, not pure white) become “black”
- The e-paper negates black and white

... Maybe not accurate, I'm new to such thing 🙃

Just mention that, it's totally OK to serve a Cat-Printer instance on PC & access it with phone browser. You may debug in such way, try to change a bit CSS 😉
(Will use PC Bluetooth to print, though)

frankenstein91, 2022-09-03:
[image]
I thought you can switch between light and dark with the Dark Theme button.

NaitLee, 2022-09-03:
Uh, well, it's always dark when you have *High Contrast* enabled.

CSS isn’t flexible enough at these, so I left it as that.
(see all the existing duplicate rules in `www/main.css`, notably between `@media` and `body.*`, already annoying indeed)

Maybe use JS? I don't like that feeling anyway.

Thoughts are welcome :)