NaitLee/Cat-Printer #39: [Add Device] YHK-0095

moriel5, 2022-09-08:
It would appear that these cat thermal printers can also arrive as "YHK-0095", as my friend's did.

Here are the device specifications, obtained via NRF Connect.

Client
**Generic Access**
UUID: 0x1800
PRIMARY SERVICE

- **Device Name**
- UUID: 0x2A00
- Value: YHK-0095

**Unknown Service**
UUID: 49535343-fe7d-4ae5-8fa9-9fafd205e455
PRIMARY SERVICE

- **Unknown Characteristic**
- UUID: 49535343-1e4d-4bd9-ba61-23c647249616
- Properties: NOTIFY
- **Descriptors:**
- UUID: 0x2902
- Value: Notifications enabled

- **Unknown Characteristic**
- UUID: 49535343-6daa-4d02-abf6-19569aca69fe
- Properties: WRITE

- **Unknown Characteristic**
- UUID: 49535343-8841-43f4-a8d4-ecbe34729bb3
- Properties: WRITE, WRITE NO RESPONSE

- **Unknown Characteristic**
- UUID: 49535343-aca3-481c-91ec-d85e28a60318
- Properties: NOTIFY, WRITE
- **Descriptors:**
- UUID: 0x2902
- Value: Notifications enabled

Server
**Generic Attribute**
UUID: 0x1801
PRIMARY SERVICE

- **Service Changed**
- UUID: 0x2A05
- Properties: INDICATE

**Generic Access**
UUID: 0x1800
PRIMARY SERVICE

**Device Name**
- UUID: 0x2A00
- Properties: READ

- **Appearance**
- UUID: 0x2A01
- Properties: READ

- **Central Address Resolution**
- UUID: 0x2AA6
- Properties: READ

NaitLee, 2022-09-08:
The newest version 0.6.0.2 can help you to test if this (unknown) model is directly supported, tell me if it works.

I will see if these Bluetooth specs are the same/similar as mime at later time.

(Brands are gradually changing. As time goes, those printers get less chance of being supported — unless we get to know how newer printers work)

moriel5, 2022-09-08:
Thanks, this is the error I am getting:
Error: API Failure
    at http://127.0.0.1:8095/~every.js:740:28
    at async callApi (http://127.0.0.1:8095/~every.js:729:12)
{
    "name": "Exception",
    "details": "'YHK-0095'"
}

moriel5, 2022-09-08:
If this helps (it probably doesn't), the chip is a JL ac21bp09674.

I was unable to find any information about it.

The chip is a standard SPI chip (I forget whether it is SOP-8 or WSON-8), so if this helps out, I can dump it's firmware with my CH341b programmer (the thermal  printer is still at my place, since my friend is currently here).

moriel5, 2022-09-08:
The hardware:

Chipset: JL AC21BP09674
Power IC: Chipstar CS5090E
Motor Driver: Global Mixed-mode Technology G2053
SPI(?): Unbranded 4453

NaitLee, 2022-09-08:
Ouch, it will be tough.

The “extreme” job I had done is to dump traffic, with NRF Connect faking the printer & communicate with original (proprietary) software.
That means 2 phones, manual configuration with your specs, temporary compromise of freedom...
You may also want to “print” kind of special image, for producing good dump data — checkerboard is great for this, try to make 16px checkers with GIMP.
Typical cat printers have printed content width of 384 pixels. Try to know if it’s true in your case.

But it may be difficult to work with, if printer also transmits data to software for e.g. confirmation.
So if that doesn’t work either, may try to RE. Tell me its name.

(I don’t think I have ability to deal with firmware)
(I shall rest, will be offline for some hours)

moriel5, 2022-09-09:
I had created the image (in both .png and .jpg, just in case one of the formats isn't supported), however it appears that the thermal printer wasn't completely unarmed, as the power management IC was also burnt, so by the time the images were ready, the battery was dead, however we are unable to recharge it.

For the time being, I had ordered a replacement CS5090E off AliExpress, however we cannot run any tests until it arrives, at which point I will desolder the old chip and solder on it's replacement, however.

We also ordered another thermal printer, since my friend wanted a second one in any case, and we really liked the Milestone MHT-P11 (a business-oriented thermal printer), since it has USB-C rather than microUSB, and can be used with desktop operating systems (Windows, MacOS, standard Linux, etc.), both via Bluetooth and USB. Plus, with discounts, it became the cheapest option.

Something really cool about the company, is the fact that they have SDKs available, for at least some of their models, so it may be possible to add support for it more quickly and easily.

NaitLee, 2022-09-09:
I used to power up some devices that need battery with direct 3.7 V power.
For the printer it may be 3.3 V. I don’t know if it will work without power management module.

In [this repo](https://github.com/JJJollyjim/catprinter) are some disassembled samples & specs of *typical* “cat printers”. I guess we are facing a totally different one.

Back to the time when I need to know more about the protocol, I bought an expensive programmable NRF Bluetooth chip, which requires a large (> 200 MiB), heavy (Electron based, expect what) & stupid (want to be root, always segfault) SDK to work, and for its closed/proprietary nature the experience is extremely unpleasant.
My mind tells me “everything that requires an SDK sucks”. I want to see something simple to be understood — now there’s Cat-Printer, the protocol is in `printer_lib/commander.py`, simple like that.

For reality reason I have little and little time to contribute, I will try my best. Good luck!

moriel5, 2022-09-09:
Thanks, for the information and goodwill, once my friend's thermal printer is up and running, I'll make sure to do what I can to help as well.

The battery is a dual battery pack, so 2 x 3.7V, the printer works even without functioning power management, however since it did not detect the battery, it always outputs the battery at 100%, and is unable to charge.

Regarding the pictures shown there, my friend's unit looks exactly the same as the GB02 and the GT01, except with a different main chip, of course (the other chips are interestingly not mentioned in the other reports).

Not the PineTime? That is a good and relatively cheap development device for Nordic's NRF platform.

And regarding SDKs, I only meant that it could help lessen a bit with reverse engineering, since you could theoretically understand some of what is going on from the documentation (although legal issues could arise with proprietary SDKs), I fully agree with your kind of thought, however. They is why I initially (not knowing the extent of this situation) recommended my friend specifically one of these "cat printers", because of your hard work and impressive software (GUI could be improved a little, however the general direction is pretty nailed down, so any GUI changes would be very minor).

moriel5, 2022-09-11:
A bit late, however here are the pictures I took back then:

[image]

[image]

NaitLee, 2022-09-11:
A fact is that I didn’t ever disassemble my printer.
The ability to hack around hardware and even firmware is cool.
Sadly I hardly get disassembled things put back together again. (Mostly because what I disassemble is already broken)
I don’t think buying another printer is a choice for me — except when a friend also want one.
Good luck!

I have a random thought, you can try to replace the `tx_characteristic` inside `printer.py` to one of your printer’s (with `write` property), restart application and try print to see if something will happen.

```python
    tx_characteristic = '0000ae01-0000-1000-8000-00805f9b34fb'
    rx_characteristic = '0000ae02-0000-1000-8000-00805f9b34fb'
```

I see there’s one identical characteristic uuid being in your printer and typical cat printers, though not being used explicitly. Sometimes coincidence just works.

moriel5, 2022-09-11:
Thanks, I'll try that after I repair my friend's thermal printer (so that we can charge it, it would take the same time to get a charger to charge the battery pack directly).

The Cat printers are actually very easy to disassemble without breaking anything, from what I saw. You should try it too.

I generally disassemble electronics they I purchase as soon as I can (I literally disassembled my Brocade FCX-648S-HPOE managed enterprise network switch first thing when it arrived at the doorstep).

RaspberryPiDude314, 2022-09-26:
I second this. I have a YHK-EE77 that looks pretty much identical to the above photos and probably works the same internally, and would love to get it running through code.

moriel5, 2022-10-18:
Hopefully I'll be back with information next week, I still need to pick up the power ICs and solder them.

In the meantime, I'm opening a new issue regarding the Milestone MHT-P11.

moriel5, 2023-08-28:
Ah, that is certainly unfortunate.

But since Bluetooth profiles are software based, perhaps someone will eventually build a new firmware for that does support Bluetooth LE and the GATT protocol.

For now, I guess that this issue will need to be closed, but at least there is some hope for the future.

Cris-lml007, 2024-06-09:
> I wandered around the internet and met this: https://github.com/abhigkar/YHK-Cat-Thermal-Printer
> 
> It contains protocol for YHK-* printers.
> 
> The author proved that YHK printers use classic bluetooth, which isn’t architecture-compatible with this project. (i.e. better for a dedicated project)
> 
> I don’t think I could get this brand in hand. So maybe it’s left to other guys to make an actual application…

thanks to this thread, I managed to get my mini cat printer (YHK) working, I saw that the repository you mention does not have all the support for said printer, and I decided to create a new [repository](https://github.com/Cris-lml007/YHK-miniPrinter) with those bases.  
support for cups is still missing, if anyone has knowledge of this, I would greatly appreciate it if you could collaborate.

venthewolf, 2026-10-11:
Thanks to @Cris-lml007 script i got mine to work. So its resolved