NaitLee/Cat-Printer #41: Android app: can't scan devices

pferreir, 2022-09-18:
Every time I try to scan a device I get an error:

```
Error: API Failure  
at handleBluetoothProblem (http://127.0.0.1:8095/~every.js:1475:17)  
at async http://127.0.0.1:8095/~every.js:739:45  
at async callApi (http://127.0.0.1:8095/~every.js:729:12)  
at async Main.searchDevices (http://127.0.0.1:8095/~every.js:1482:29)  
{  
"name": "Exception",  
"details": "JVM exception occurred: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@3d8431d6: GattService registerScanner java.lang.SecurityException"  
}
```
If I try again:
```
Error: API Failure  
at handleBluetoothProblem (http://127.0.0.1:8095/~every.js:1475:17)  
at async http://127.0.0.1:8095/~every.js:739:45  
at async callApi (http://127.0.0.1:8095/~every.js:729:12)  
at async Main.searchDevices (http://127.0.0.1:8095/~every.js:1482:29)  
{  
"name": "BleakError",  
"details": "A BleakScanner is already scanning on this adapter."  
}
```

The app has permissions "Location" and "Nearby Devices" set to "Allowed". I am on Android 13.

NaitLee, 2022-09-18:
Sadly I can’t gather an Android 13 phone to give a test.

`BLUETOOTH_SCAN` is defined in build parameters, and with built app my phone (Android 11) have no problem.

Double check if the basic Bluetooth permission was given. Maybe also to check if it’s shown in permission details when installing.
Which version are you using: `0.6.1` or `0.6.0.2`? There are technical differences between them, so try the other version if one isn’t right.

parkerlreed, 2022-09-18:
I'm on Android 12 on a Samsung Note 10 and seeing the same thing. Location and nearby devices is enabled in permissions but the first scan says it doesn't have permissions and then when you try to do it again it says there's already a scanning thread going on.

parkerlreed, 2022-09-18:
Same on 0.6.0.2

[image]
[image]

parkerlreed, 2022-09-18:
Here's a thread on some possible extra permissions that might be needed https://stackoverflow.com/questions/67722950/android-12-new-bluetooth-permissions

NaitLee, 2022-10-01:
Sorry for being late. I just got time to test on an Android 12 device and reproduced it.

Obviously there should be extra code for requesting permission on Android 12+, as said in your link.

The code in theory could be in `server.py`, using python-for-android `android` or `pyjnius` package.

But the painful part is, I really have no idea to know where those Java namespaces/classes are coming from.
(I’m wondering if asking in Stack Overflow will help)

The code could also be in the Java files (mentioned in the Android build env instruction), but I’m tired about them :(

(It’s a bit holiday for me, without guarantee to do anything, for life matters being messy)

NaitLee, 2023-02-12:
Didn’t notice this closed automatically. Try the new release 0.6.2.0 and see if it’s solved?

sync1211, 2023-02-12:
0.6.2.0 does not seem to work for me on Android 12.
I'm going to try and rebuild it using the latest commit to see if it might be a dependency issue.

sync1211, 2023-02-12:
The app still works when I compile it using the latest commit.

I am using p4a Version 2023.01.28, which does not support `--orientation user` (I've used `--orientation portait` for builds).
I'm using NDK version r25b with an API target of 30 (Build says "min API 21").

The only real with my build process is that I've downloaded bleak from PyPi, instead of from GitHub. (missing `Setup.py`)

I am building the app on Manjaro 22.0.2 without the docker environment.

NaitLee, 2023-02-12:
I’ve upgraded p4a (2023.02.10) and ndk (r25b) to build this one:
[cat-printer-android-0.6.2.1.apk.zip](https://github.com/NaitLee/Cat-Printer/files/10716382/cat-printer-android-0.6.2.1.apk.zip)
(I don’t know if bleak is updated)
it’s a release build, you may remove the zip suffix and install as apk. If it’s ok then let’s release it.
(hardly see android 12+ around me)

pferreir, 2023-02-12:
Hi there, thanks for the update! I've just tested it and unfortunately it
doesn't seem to work yet. I get the same thing about needing "Bluetooth"
permission.

NaitLee, 2023-02-12:
I cleaned up by build environment but can’t put it back anymore. Until some day I’m lucky enough…

@sync1211 you may put your build here if you want :)

sync1211, 2023-02-12:
I haven't been able to sign my builds yet, so I only have a debug build atm:
[cat-printer-debug-0.6.2.0.zip](https://github.com/NaitLee/Cat-Printer/files/10716643/cat-printer-debug-0.6.2.0.zip)

Cratesmith, 2023-02-13:
Can confirm that the release doesn't work on my phone/mx06 but @sync1211's debug build did.

sync1211, 2023-02-17:
I've managed to replicate the build using an archlinux docker environment.

The steps are almost the same as the [manual-build guide](https://github.com/NaitLee/Cat-Printer/blob/main/build-android/manual-steps.md).
You can find my notes for building within docker [in this Gist](https://gist.github.com/sync1211/ee92c90f74c27aace0f0ac9113b8a891).

NaitLee, 2023-04-08:
FINALLY I got the environment to work. Here’s a debug build with almost all of the newest things: [cat-printer-debug-0.6.2.1.apk.zip](https://github.com/NaitLee/Cat-Printer/files/11183690/cat-printer-debug-0.6.2.1.apk.zip).
(remove the `.zip` suffix and install it)

Please give it a test!

sync1211, 2023-04-08:
I just installed it and managed to scan and print successfully.

NaitLee, 2023-04-09:
Thanks to all of you, there’s 0.6.2.1 😄

NaitLee, 2023-04-17:
No problems so far, closing this gracefully 🎉

PR #53 (2023-02-11): Add Bluetooth permission request for Android 12+
This PR adds a permission request for `BLUETOOTH_SCAN` and `BLUETOOTH_CONNECT`.
This permission request is required to scan for devices on Android 12 or higher.

Fixes #41
[image]
closed 2023-04-17 by NaitLee