NaitLee/Cat-Printer #42: GB03 darkness virtually unaffected

parkerlreed, 2022-09-21:
Image examples. The last image is the same settings as the next to last but with quality all the way down and then all the way up. It slightly changes the sound but not the output. 

Default Android application (iPrint) darkens as expected so the hardware is fine. 

Using the python server on Arch Linux. 

[image]
[image]
[image]
[image]

parkerlreed, 2022-09-21:
If I print from command line it's perfect...

```
(64)(deck@steamdeck Cat-Printer)$ python printer.py -c image ~/Downloads/Tux.png 
Cat Printer
Connecting
Finished
Disconnecting from printer
```

parkerlreed, 2022-09-21:
Think I narrowed this down. There is no restriction to the "energy" value when setting is via "Strength" slider in the web interface.

If you move off of the default (like drag it higher) it caps out at 256 which is an invalid value for the printer, this printing everything very light. I set it back to 64 in config.json and didn't touch "Strength" slider and it printed fine.

parkerlreed, 2022-09-21:
From index.html might make more sense to cap it at the max of 96 for text

```
                        <label class="label-span-input" data-hide-as="print">
                            <span data-i18n="strength-">Strength:</span>
                            <input type="range" min="0" max="256" value="64" step="16" name="energy" data-key data-default />
```

NaitLee, 2022-09-22:
I remember the last change to `server.py`  changed default settings from 0.2 to an other value, for being really an integer -- but didn't change the "factor".
So it's possibly that, bigger value (> 96?) being multiplied (by factor) and excessed `0xffff`, being truncated to a small value, then printed very light.
It's adviced to check `server.py` instead.

(I'll hardly have coding time from now on)

sync1211, 2022-10-07:
I've been playing around with that setting for some time and it appears that falls back to the default value just after 208.

Can anyone confirm whether it's only limited to the GB03 or if it also affects the other models?

NaitLee, 2022-10-07:
The newest commit have a fixing attempt to this, adjusted where those “factors” appeared (they are poorly organized before, I admit)

Please test to prove it’s working or not. Remove the config file if necessary.
(little tip: right-click on “Settings -> Exit” will prompt if you want to reset config)

closed 2023-02-12 by NaitLee