NaitLee/Cat-Printer #45: New option for printing labels

Jardal, 2022-10-16:
Hi, would you like to add option for retract paper before another print for label printing, please?
Note, that retract step size is bigger than feed step size (about 2x more) and exact position need to be manually calibrated on up to 4 sequential prints.

File printer.py, line about 440 (in version 0.6.0.2), two functions
    def _prepare(self):
        self.get_device_state()
        **self.set_speed(48)     # set speed before retract
        self.retract_paper(51) # ADD OPTION + VALUE: retract amount for current label before print starts**
        if self.model.is_new_kind:
            self.start_printing_new()
            
........
    def _finish(self):
        self.end_lattice()
        self.set_speed(8)
        **self.feed_paper(128)  # ADD OPTION VALUE TO FEED AMOUNT**
        self.get_device_state()
        self.flush()
.......                        
            
Many thanks!

NaitLee, 2022-10-16:
Sorry but I have no knowledge about “label printing”, for my printer have no such concept.

Adding these options (to cmdline and/or web) should be relatively easy, but I really want to know what does “label printing” does 😁
It may help to optimize the feature addition, to be friendly enough.

Note that retracting of paper isn’t (horizontally) precious enough, though.

(As usual, I almost can’t do code in this period)

Jardal, 2022-10-16:
Hi, labels means stickers, a vax tape with paper slices, like here:
https://www.papirnisvet.com/proizvod/nalepnice-za-deklaracije-u-rolni-58x43mm-thermal-eco

I do it with ordinary cat printer GB-02, much better fun than with paper.

It (do retract), prints, eject label, you peel it off, and for 
continuing it need to be label at right position again, so it need to be 
retracted to another sticker position just before printing.

NaitLee napsal(a):

NaitLee, 2022-10-20:
Got it. Labels are going to be useful to me as well.

I still need info about what’s the point to retract, and how to feed/retract to not offset labels’ position.
(It may be dependent to label size. Something to calculate units (to use in `self.feed_paper`) from paper specs (w*h in millimeters) should be helpful. Maybe these have something to do with printer DPI.)

Is relative behavior available in “official” app?

NaitLee, 2022-10-23:
I bought a very large roll of labels (unbelievably, ￥7 (≈$1) for 1050 pieces, the roll is even bigger than the printer)

I guess you are feeding labels from the outside, so that retracting is required?

---

That means we need some specs to be manually specified:

- Paper width, in millimeters, to be used to calculate a ratio (to 384px). Typically 50mm.
- Label height, mine is 35mm, but it really varies by requirements.
- Gap height, that between two labels. Mine is 2mm.

I think someday I will work on this, to make full use of labels 🙂

Jardal, 2022-10-23:
I was cut length about 2 meters from the roll and make tiny roll just 
for printer. But before any inserting, the roll need to be very 
tightened to much smaller diameter that is in the printer allowed. When 
you leave fingers from it, it will quickly un-wind and block stepper 
motor from move. You need to be faster than roll.

I think that "milimeter" sizes are only as guide, not exact and after 
few prints label position will not be right. Pixels size is much exact. 
Mine labels are 50x30mm (384x238px), gap about 3mm, so my code values 
are just for it. Not calculated, just measured by printing.

Process:
- retract (from previous feed out minus gap)
- print
- feed out label to take it off

When inserting, top edge of sticker are on edge of printer, it is help 
guide for inserting new roll.

When calibrating, use cheap paper roll instead of stickers.

NaitLee napsal(a):

moriel5, 2022-10-30:
@NaitLee Hmm... might you know a source for thermal rolls that are not made from Bisphenol-based compounds (such as Bisphenol A or Bisphenol S), but rather from Urea?

I'm considering I'm the relatively near future to get myself a thermal printer as well (same as the second one my friend purchased, since I intend to start a business), however I do not want to use the standard thermal rolls due to the known health hazards associated with Bisphenol-based compounds. And as a bonus, Urea-based thermal paper is also supposed to produce higher quality prints.

The only drawback to using urea-based thermal is price, which is supposed to be higher than Bisphenol-based thermal paper.

h3ndrik, 2023-11-20:
> a source for thermal rolls that are not made from Bisphenol-based compounds (such as Bisphenol A or Bisphenol S), but rather from Urea?

@moriel5 Did you ever find a source that is trustworthy?

moriel5, 2023-11-21:
Unfortunately, nothing that I am able to order from, though there is a European provider (B2B only) that seems to be reputable (I forget the name, I'll post when I find it again).

h3ndrik, 2023-11-21:
Thx. Yeah I live in Europe. But I got one of the cat mini printers, so it takes the 57mm rolls. Might be wider if it's business supply.

moriel5, 2023-11-21:
I remember that they have a wide assortment of sizes, including 57mm, which makes sense, since businesses can be anything from offices to factories to transportation (and thermal paper is often used for receipts).