NaitLee/Cat-Printer #47: MX06 printer?

Hoshnasi, 2022-10-19:
Hello,

I have an MX06 cat printer and I've tried a few platforms (MacOS, Windows) and I can't seem to get it to run.  Is this printer model supported?

Thanks!

Cratesmith, 2022-10-20:
I have the same model and have found that the official app as well as the RawBT android app can print in it.

Is there any way to capture the info from these to add them to Cat-Printer?

Sent with [Proton Mail](https://proton.me/) secure email.

NaitLee, 2022-10-20:
Unseen models are not in this program’s base, but nevertheless with newer version you can test an unknown device. Find the feature in Settings in Web UI.

With luck it can just work. Tell me if it works, if true then we add this model to base 🙂

That said, I have no ability to inspect printer protocols, for insufficient time.

You can find our previous experience in #8 and #23. Not that organized, sorry.

I really hope a volunteer can take the job — just like how other voluntary projects are developed.
It isn’t a simple circumstance 🙁

Cratesmith, 2022-10-20:
Yeah already tested with unkown device and currently it doesn't work.
However RawBT will print from it via it's "iPrint app printers" mode.

So maybe there's a way to inspect the instructions it's sending and add those to cat-printer?

Sent with [Proton Mail](https://proton.me/) secure email.

NaitLee, 2022-10-20:
I suspected the naming “MX06” is of iPrint too. I think I could look into its internal data and find its specs, just like how I got started.

The problem is that I didn’t keep those data. To redo that I really need spare time — but I don’t have.

Another approach is to dump bluetooth traffic between iPrint and printer — the application “nRF Connect” can mimic a BLE device for you (in our case, that printer), and dump traffic.
For iPrint traffic is just similar, it may be easier in this case, since we already have most of its protocol.

I remember I have left some guidance about traffic dump here, but forgot where exactly are them. Sorry again.

Cratesmith, 2022-10-20:
All good. I'll have a go at this if I have time.

Sent with [Proton Mail](https://proton.me/) secure email.

NaitLee, 2022-10-20:
I just noted that “RawBT” app you said.
I didn’t know about it until now. But I’m wondering what mysterious power gives it ability to be compatible with that large amount of printers (and various protocols).

Cratesmith, 2022-10-20:
I'm unsure of that myself. But they seem to have very broad compatibility, not just within iPrint types.

That said the iPrint 'driver' seems to be capable of working with a broad range of devices so maybe there's a bruteforce approach that works with many of them?

Sent with [Proton Mail](https://proton.me/) secure email.

wackoracoon, 2023-02-01:
Hello,
I've a MX05 printer and I just add a new line in the models.py file and it works !
I've tried both True and False value on the boolean is_new and both works.

NaitLee, 2023-02-12:
0.6.2.0 have MX05 added.

MX06 too, but don’t know if it will work.

lazycouchpotato, 2023-02-13:
Can confirm that it is working with MX06. 

Prints need more padding though. There's a 0.8cm distance between the print head and the blade from where to tear the paper from, causing all prints to get cut off. I've added some images to display the issue. I think the ideal situation would be to have equal padding on the top and bottom, so maybe 1.6cm extra?

Also, there's a high pitch whining sound coming out of my MX06 after completing a print with Cat Printer or RawBT. There's no sound when using the device's recommended app [Fun Print](https://play.google.com/store/apps/details?id=com.fun.mxw&gl=US).

Please let me know if this should be posted as a new issue.

Thanks for all the great work!

[image]
[image]
[image]

wahlatlas, 2023-03-02:
Can confirm too that it is working with MX06 and that padding is an issue. The latter can be added to the image as a workaround in imagemagick.

`convert test.png -rotate 90 -interpolate Integer -filter point -resize 384x576 -gravity south -splice 0x150 test.pbm`

nodlek-ctrl, 2023-04-01:
I have a MX05 and padding is an issue here too. Perhaps rename the issue or start another dedicated to the padding issue?

NaitLee, 2023-04-02:
I’ve found a MX06 vendor locally. I’m still considering whether to buy it and what to do besides of testing with it.

Before that, may helpful to try to remove [this line](https://github.com/NaitLee/Cat-Printer/blob/49b2a9a0089ef3c6cd2bff86d710632817ad8f20/printer.py#L460):
```python
        self.set_speed(8)
```
re-power the printer and see if it will help to padding problem?

nodlek-ctrl, 2023-04-02:
I can try that on my MX05 in about an hour or so. From what I have seen, both models are suffering the same problem.

nodlek-ctrl, 2023-04-02:
alright.  I have tested removing the line and it does nothing. Why is it that when changing the `self.feed_paper()` value, there is no change, except when set to 256? is there something I am missing to do with python in general?

The distance between printhead and message with 256 set is 1 1/4 inch,
The distance between print head and message with 128 set is 1/2 inch.

Why is it that when set to something in between, such as 200, it is still 1/2 an inch, and not a full inch?

NaitLee, 2023-04-02:
I feel that 128 `0x80` and 256 `0x01 0x00` may make a difference to that printer. You may try some other values, like `1`, `32`, `255`, `4096` or even `32768` (just power it off if it’s feeding too much 😁)

I coded for a temporary solution: mark model MX05/MX06 as problematic on feeding and let them print some empty lines instead of giving feed command. This is committed to the main branch, you may want to check it out.

The value is 128 for a long time, which should be okay. Probably it should be adjustable in the future.

nodlek-ctrl, 2023-04-03:
That did the trick. Guess it is the way the printer handles feeding paper.

NaitLee, 2023-04-08:
I have gifted my friend a MX06 printer :)

If no problem’s left and we got next Android release to work, we may close this.

huaak, 2023-04-15:
When inserting MX08 (sold as "C15") in models.py in models List in line 22 and in line 30 for the mx05/06-workaround, it prints nicely too.. :)

NaitLee, 2023-04-16:
> When inserting MX08 (sold as "C15") in models.py in models List in line 22 and in line 30 for the mx05/06-workaround, it prints nicely too.. :)

0.6.3.0 release have it added :)

huaak, 2023-04-16:
..here an update for german language file (I just translated the new
english phrases lines 145ff..):

     "pf2-font-not-found-or-broken-0": "PF2 font nicht gefunden oder
defekt: '{0}'",
     "try-to-print-through-an-unknown-device": "Versuch mit unbekanntem
Gerät zu drucken",
     "scanning-for-all-bluetooth-devices-nearby": "Scanne nach allen
Bluertooth-Geräten in der Nähe…",
     "there-are-multiple-devices-": "Mehrere Geräte gefunden:",
     "choose-which-one-0-": "Geräteauswahl? [{0}]: ",
     "multiple-devices-found-please-specify-one": "Mehrere Geräte
gefunden, bitte Auswahl treffen",
     "no-prompt-for-multiple-devices": "keine Aufforderung für mehrere
Geräte"
}

Also I did a first attempt to create a nl-NL languagefile to support
dutch (nederlands) attached too..

Greetings from germany,

Huaak

NaitLee, 2023-04-16:
Thanks @huaak! 🎉

> Also I did a first attempt to create a nl-NL languagefile to support dutch (nederlands) attached too..

Seems there’s no attachment around here, probably because of by email.
So may consider uploading it by GitHub interface?

Usual practice of such contribution is to Fork a repo, clone & work on then Merge it, which keeps the “state” — the Internet may give you introduction :)

huaak, 2023-04-16:
..okay, I never did it before, but I tried it..hope it works!?.. ;)

NaitLee, 2023-04-17:
Well done! 🎉

closed 2023-04-17 by NaitLee