NaitLee/Cat-Printer #48: versioncodes messed up

IzzySoft, 2022-10-23:
Please take care to always increase (and never decrease) `versionCode`, as that's what Android uses internally to decide whether an APK is an update to the already installed app or not. Currently, there's an issue:

* v0.6.0.1 used `versioncode` 12160001
* v0.6.0.2 used `versioncode` 12160002
* now v6.0.1 uses `1021601`, which is lower (1021601 < 12160002) – so users having one of the two previous versions installed will never receive an update.

Al future versions must have a `versionCode > 12160002` to be accepted by existing installations. Can you please take care for that? Thanks in advance!

NaitLee, 2023-02-12:
Sorry for not replying earlier. The future versions will always have version code like `0.6.2.0`

(Shouldn’t it be Python for Android to blame? 🙃 It’s really increasing a minor version)

IzzySoft, 2023-02-12:
> The future versions will always have version code like `0.6.2.0`

That's the `versionName`, which is just for humans to read. For Android, `versionCode` (an integer) is the important one (I guess if that's based on `versionName` it would be fixed as a side-effect, so all would be fine then). And no idea which tool to blame – I didn't intend "blame", just trying to help :smile:

NaitLee, 2023-02-12:
Yes I know 😄 I’m not strictly android dev and this project is just using python-for-android to do those works. Things like version “code” are out of my control, and p4a just wanted a version name.

So as said versioning should be ok from now on.

closed 2023-02-13 by NaitLee