NaitLee/Cat-Printer #49: Problems building apk

bg16-2009, 2022-11-19:
When I try building the app I get this error after running the last command from [this tutorial](https://github.com/NaitLee/Cat-Printer/blob/main/build-android/manual-steps.md):
[image]

sync1211, 2023-02-10:
What's the directory you're using for your build? Are you using a docker container or building directly on your host system?

bg16-2009, 2023-03-13:
I was using a docker container

NaitLee, 2023-04-08:
Newer versions of bleak have no `setup.py` included for setting up as it’s a deprecated way, but `python-for-android` still needs it for a recipe.
(Thanks to [this comment](https://github.com/NaitLee/Cat-Printer/issues/41#issuecomment-1427044168))

You can find an older version of bleak at [here](https://pypi.org/project/bleak/0.19.5/#files), extract it and use it in place of bleak git repository.

Sorry for being so late. I don’t know if old versions will keep working in the far future, either.

closed 2025-05-02 by bg16-2009