NaitLee/Cat-Printer #51: Rearrange build infrastructure

NaitLee, 2023-02-07:
From today’s view it’s messed up. I hope this can be done to make it better and address some problems like #48.

sync1211, 2023-02-10:
I recently started porting the build scripts to Makefile as it allows for greater customization. ([Makefile-Branch](https://github.com/sync1211/Cat-Printer/tree/Makefile))

To create a bundle, the user would only need to run `make`, instead of `0-prepare.sh && cd build-common && python build.py $version`

NaitLee, 2023-02-12:
I remember a long time ago there is #16. I didn’t yet merge it.

One existing problem is Windows building is broken. Even if it’s fixed, it’s going to break again later time.

Python is unfriendly at this… you can’t get `bleak-winrt` via pip on *nix, but need to manually take the `.whl` file from pypi (it’s a zip file), then do some dirty work embedding to python, then *test it on a real Windows machine* — which became unacceptable to be by time.