NaitLee/Cat-Printer #56: Cant open main.py and (or just) server.py

Eduardfabian21, 2023-03-14:
When ever i try to open it says this
[image]
Im using:
Windows 10 Home version 22H2
Latest version of Python
Latest version of Cat Printer

Eduardfabian21, 2023-03-14:
+ i forgot to say that i also tryed bare one

NaitLee, 2023-03-15:
Are you sure it’s the latest version of Python? Try `python3 --version` to see the interpreter version.

By the way, try if this snippet will work:

```python
a = 10
while a := a - 1:
    print(a)
```

The `:=` operator is added in Python 3.8, it should be earlier versions showing a Syntax Error.

Eduardfabian21, 2023-03-15:
To my surprize, it outputed: `Python 3.7.9`

NaitLee, 2023-03-15:
As far as I can remember, on Windows, typical command to invoke Python is `py`.

I don’t know about old versions, probably an old version was kept for `python3` in your case (and probably by something else.)

Eduardfabian21, 2023-03-15:
I tryed with 3.10.10 and it says:
`Traceback (most recent call last):`
`  File "C:\Users\Edi\Downloads\cat-printer-pure-0.6.2.0\cat-printer\main.py", line 3, in <module>`
`    from server import serve`
`  File "C:\Users\Edi\Downloads\cat-printer-pure-0.6.2.0\cat-printer\server.py", line 22, in <module>`
`    from printer import PrinterDriver, PrinterError, i18n, info`
`  File "C:\Users\Edi\Downloads\cat-printer-pure-0.6.2.0\cat-printer\printer.py", line 78, in <module>`
`    raise error`
`  File "C:\Users\Edi\Downloads\cat-printer-pure-0.6.2.0\cat-printer\printer.py", line 74, in <module>`
`    from bleak import BleakClient, BleakScanner`
`  File "C:\Users\Edi\Downloads\cat-printer-pure-0.6.2.0\cat-printer\bleak\__init__.py", line 73, in <module>`
`    from bleak.backends.winrt.scanner import (`
`  File "C:\Users\Edi\Downloads\cat-printer-pure-0.6.2.0\cat-printer\bleak\backends\winrt\scanner.py", line 6, in <module>`
`    from bleak_winrt.windows.devices.bluetooth.advertisement import (`
`ModuleNotFoundError: No module named 'bleak_winrt'`

Eduardfabian21, 2023-03-15:
And on windows you can still use python3 cmd EVEN if its a newer version

Eduardfabian21, 2023-03-15:
If i use py main.py, it opens a new windows

Eduardfabian21, 2023-03-15:
and i forgot to say that, windows forces me to download it from the windows store

NaitLee, 2023-03-15:
btw `main.py` is unimportant, it’s just a helper script for Android and does nothing special other than “executing” `server.py` by importing.

AndCouvel, 2023-03-15:
I have Python 3.11.2 🥲

Eduardfabian21, 2023-03-15:
It outputed:

Requirement already satisfied: bleak in c:\users\edi\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (0.19.5)
Requirement already satisfied: bleak-winrt<2.0.0,>=1.2.0 in c:\users\edi\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from bleak) (1.2.0)
Requirement already satisfied: async-timeout<5,>=3.0.0 in c:\users\edi\appdata\local\packages\pythonsoftwarefoundation.python.3.10_qbz5n2kfra8p0\localcache\local-packages\python310\site-packages (from bleak) (4.0.2)

Eduardfabian21, 2023-03-15:
I also saw that you still need python for the older ver of cat-printer
[image]

NaitLee, 2023-03-15:
Seems the pure bundle have outdated dependency.

You may want the bare bundle, or simply clone the repository.

That batch script is intended for a complete Windows bundle (available for some earlier versions), where an embeddable Python 3 is included (and that’s named `python.exe`, hence `python`)

NaitLee, 2023-03-15:
Hi @AndCouvel, I didn’t see Python 3.11.2 problems here. You may send some snapshots to show what’s exactly going on your side :)

AndCouvel, 2023-03-18:
> Hi @AndCouvel, I didn’t see Python 3.11.2 problems here. You may send some snapshots to show what’s exactly going on your side :)

  andrescouvel@Macbook-Air-de-Andy cat-printer % python3 server.py
Traceback (most recent call last):
  File "/Users/andrescouvel/Downloads/cat-printer/server.py", line 22, in <module>
    from printer import PrinterDriver, PrinterError, i18n, info
  File "/Users/andrescouvel/Downloads/cat-printer/printer.py", line 48, in <module>
    i18n = I18nLib(path).translate
           ^^^^^^^^^^^^^
  File "/Users/andrescouvel/Downloads/cat-printer/printer_lib/i18n.py", line 23, in __init__
    self.lang = lang or (locale.getdefaultlocale()[0] or fallback).replace('_', '-')
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'replace'
andrescouvel@Macbook-Air-de-Andy cat-printer % 

 that is the error code, running on Macbook Air M1

NaitLee, 2023-08-27:
In case technology is a problem, just use this new Web app: https://print.unseen-site.fun/

Be sure to use a Chromium-based browser for it. Safari is probably ok too.