NaitLee/Cat-Printer #57: Not running on MacBook M1

AndCouvel, 2023-03-18:
I downloaded the resources and the bare version, and this is the output:

`andrescouvel@Macbook-Air-de-Andy Cat-Printer-0.6.2.0 % cd /Users/andrescouvel/Downloads/cat-printer-2 
andrescouvel@Macbook-Air-de-Andy cat-printer-2 % python3 server.py                              
Traceback (most recent call last):
  File "/Users/andrescouvel/Downloads/cat-printer-2/server.py", line 22, in <module>
    from printer import PrinterDriver, PrinterError, i18n, info
  File "/Users/andrescouvel/Downloads/cat-printer-2/printer.py", line 48, in <module>
    i18n = I18nLib(path).translate
           ^^^^^^^^^^^^^
  File "/Users/andrescouvel/Downloads/cat-printer-2/printer_lib/i18n.py", line 23, in __init__
    self.lang = lang or (locale.getdefaultlocale()[0] or fallback).replace('_', '-')
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'replace'
andrescouvel@Macbook-Air-de-Andy cat-printer-2 % `

Python 3.11.2

NaitLee, 2023-03-19:
I feel it’s #38 revived, or at least something similar.

I’d try to look into it when I have some time.

NaitLee, 2023-04-03:
So that I did an incomplete fix to #38. I tried to really fix it in the newest commit.

I can’t determine what time to release a new version, but you can always clone this repository and use the newest source code as you like :)

NaitLee, 2023-04-28:
There’s a new version with fix attempt to this. Waiting for you to try it :)

jedahan, 2023-04-28:
I was able to get python server.py running on my M1 fine with latest main

krim404, 2024-02-15:
this can be closed as it works perfectly fine with M1, even cups printing.

NaitLee, 2024-02-15:
ok, closing this one

closed 2024-02-15 by NaitLee