NaitLee/Cat-Printer #58: Error "TypeError: expected str, bytes or os.PathLike object, not NoneType" on command line.

nodlek-ctrl, 2023-04-01:
Command executed- 

> [image]

Output- 

> [image]

I have tried-

- Checking and creating diary.txt (not empty)

- Power printer on and off

Most likely me doing something wrong, so some help would be fantastic.

NaitLee, 2023-04-01:
Likely the system doesn’t have ImageMagick installed.

I guess this is a Raspberry Pi, try to install it:

```bash
sudo apt install imagemagick
```

Then retry.

I think the code should give a clear solution for such case. I hereby leave a todo :)

P.S. Usually, energy efficiency is desired on a device like Raspberry Pi, so you may want to use PF2 font instead. See #10 for some info, and #29 for a real case.

nodlek-ctrl, 2023-04-01:
Thanks. I installed magick and now I get this error.
[image]

Right now, energy efficiency isn't a concern as I am not using batteries. Tanks for the tip however.

nodlek-ctrl, 2023-04-01:
Hell yeah! PF2 works a charm!
Tanks again. I might have to take a look at that todo list and get some of the easy ones done for you!

closed 2023-04-01 by nodlek-ctrl

NaitLee, 2023-04-02:
IMHO Debian/Ubuntu *always* have outdated packages served (they call it “stable” 🙂), in your case it’s [IM 6](https://legacy.imagemagick.org/) but there’s already [IM 7](https://imagemagick.org/) a long time ago.

IM have some [“policy” thing](https://imagemagick.org/script/security-policy.php) that controls what kind of thing could be converted or not, for security. It’s an XML config. Likely the default config doesn’t allow you to convert (either) from text (or) to PBM format (or both).

I forgot where it is and how to properly change it. The Internet may tell you more.

Congrats for solving your problem 🎉

NaitLee, 2023-04-02:
I think this problem is a good point to start. If you can do Python, I have some idea —

Code relative to telling ImageMagick is [here](https://github.com/NaitLee/Cat-Printer/blob/49b2a9a0089ef3c6cd2bff86d710632817ad8f20/printer.py#L561).

But it returns `None` if not found, thus the initial problem happened.

We could make it more friendly to:
- instead of just failing, tell the user that ImageMagick is not installed (or not present in `PATH` env), so try to install one
- fallback to PF2 printing

And more of it:
- PF2 font fallback. If specified font isn’t present, use another in current directory, or find one (in `/boot/grub/` with `os.walk`), or generate one (with `grub-mkfont` and fonts in `/usr/share/fonts`), or download one (from some URL)

I think I should include a small PF2 font in next releases, just in case.

nodlek-ctrl, 2023-04-02:
Alright! I can do python, I just usually struggle understanding larger scripts. 
Now that you've pointed out the important bit, I should be able to whip up a pull request sometime in the next week.

Good on you for helping people out! ⭐

commit 23d9d6d5 (2023-04-15): Merge pull request #59 from nodlek-ctrl/imagemagik-usability

Fix #58 and add some more configuration information regarding IM Security Policy.