NaitLee/Cat-Printer #62: Add priny by share option

plokko, 2023-04-18:
A very usefull addition to the android App would be adding a share option for text and images:
when you are in your favourite App, ex. Google Photo to print your photos or Google Keep to print your to-do list you could directly print what you want by sharing the image or text to the Cat-Printer App.

This should be really simple to do (by just adding intent filters and handling the image/text as it was user input) but i'm not really used to Python or p4a.

NaitLee, 2023-04-18:
Great idea 😆
Developing for Android requires a good dev environment. I’ve searched around for some tutorials, and may try it days later.
Anyway it should be really possible, by either Python/p4a or some Java code.

plokko, 2023-04-22:
I already found examples of implementations and it should be doable with few line of Python/P4A; you should find few examples on the web [like this](https://github.com/Android-for-Python/Share-Receive-Example).
 
Anyway thank you for your GREAT app: i bought the printer as a gift but it's App was unusable; i was already trying to reverse enginner the BLE communication or try ESC/POS commands with no avail.