NaitLee/Cat-Printer #65: Can't print from Windows

robisxx, 2023-05-09:
Hi.
I donwloaded 0.6.0.2 version for Windows but I can't print anything, there is no way to choose my cat printer in device list. I paired printer with my PC.
[image]
[image]

NaitLee, 2023-05-10:
The newest version is 0.6.3.0, which changed much comparing to 0.6.0.2.

Notably a Windows bundle isn’t available in new versions, because it’s broken and can’t be easily fixed.

You may want to follow README and manually setup Python and other dependencies to run new versions on Windows.

Also note that only newer versions of Windows 10 (and 11) are supported.

nfirex, 2023-07-02:
I just checked for my MX06 and it works (for Windows 10):
1) Install pyhon3 (https://www.python.org/downloads/), add python env in system.
2) Install python bleak in command line (pip install bleak)
3) Clone main branch (https://github.com/NaitLee/Cat-Printer)
4) Make start.bar in cloned directory script that will start server.py
```
@echo off
title Cat Printer - Console
python server.py
```
5) Run start.bat from command line
6) Press Scan and it will find paired printer (my case - MX06)
[image]
7) PROFIT!

@NaitLee great thanks to you! Awsome project!

supremelucifer, 2023-09-29:
> I just checked for my MX06 and it works (for Windows 10):
>
>     1. Install pyhon3 (https://www.python.org/downloads/), add python env in system.
>
>     2. Install python bleak in command line (pip [...]

The real MVP. Thnx bro!