NaitLee/Cat-Printer #71: Weak print

TheKing1248, 2023-07-31:
-Macbook Air M1
-MX06
Independente das configuracoes no painel web, a impressao sai fraca comparada com o app para iOS.

NaitLee, 2023-07-31:
As far as I know, in a long time there's an obscure bug that I'm unable to debug: the "energy" parameter may not be passed correctly from web to backend.

You may look at content of the config file of cat-printer, `config.json`, in its directory.

```jsonc
{
    "config_path": "config.json",
    "version": 4,
    // ...
    "energy": 96, // <-- HERE
    "quality": 36,
    // ...
}
```

See if this one is strange? (normally [48..255], integer)
If yes, try modify that value to e.g. `96` , save it then restart the program & refresh webpage.

LeoTech10, 2023-11-06:
I'm having the same issue, and i can confirm that changing the "energy" value to 96 fixed the issue. But with me this issue happened when i switched to different paper, with the paper i had before the default value was fine. So i guess it depends on what type of paper is used?