NaitLee/Cat-Printer #72: asyncio.exceptions.TimeoutError

M1ghty-boy, 2023-08-01:
Windows 11 23H2 25915
Python 3.8.5

Unsure if this first part is a part of the issue or my ~~cat~~ panda printer not being supported, but first off it can't find my device automatically, and when selecting it manually it spits this out after a second, and prints do not work:
```
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 61839)
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "server.py", line 331, in finish_request
    self.handler.__init__(request, client_address, self)
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 720, in __init__
    self.handle()
  File "C:\ProgramData\Anaconda3\lib\http\server.py", line 427, in handle
    self.handle_one_request()
  File "server.py", line 104, in handle_one_request
    super().handle_one_request()
  File "C:\ProgramData\Anaconda3\lib\http\server.py", line 415, in handle_one_request
    method()
  File "server.py", line 273, in do_POST
    self.handle_api()
  File "server.py", line 248, in handle_api
    self.printer.connect(name, address)
  File "D:\takeout\cat-printer-bare-0.6.3.0\cat-printer\printer.py", line 356, in connect
    self.loop(
  File "D:\takeout\cat-printer-bare-0.6.3.0\cat-printer\printer.py", line 327, in loop
    results.append(self._loop.run_until_complete(future))
  File "C:\ProgramData\Anaconda3\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\ProgramData\Anaconda3\lib\site-packages\bleak\backends\winrt\client.py", line 249, in connect
    await asyncio.wait_for(event.wait(), timeout=timeout)
  File "C:\ProgramData\Anaconda3\lib\asyncio\tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
----------------------------------------
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 61850)
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "server.py", line 331, in finish_request
    self.handler.__init__(request, client_address, self)
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 720, in __init__
    self.handle()
  File "C:\ProgramData\Anaconda3\lib\http\server.py", line 427, in handle
    self.handle_one_request()
  File "server.py", line 104, in handle_one_request
    super().handle_one_request()
  File "C:\ProgramData\Anaconda3\lib\http\server.py", line 415, in handle_one_request
    method()
  File "server.py", line 273, in do_POST
    self.handle_api()
  File "server.py", line 248, in handle_api
    self.printer.connect(name, address)
  File "D:\takeout\cat-printer-bare-0.6.3.0\cat-printer\printer.py", line 356, in connect
    self.loop(
  File "D:\takeout\cat-printer-bare-0.6.3.0\cat-printer\printer.py", line 327, in loop
    results.append(self._loop.run_until_complete(future))
  File "C:\ProgramData\Anaconda3\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\ProgramData\Anaconda3\lib\site-packages\bleak\backends\winrt\client.py", line 249, in connect
    await asyncio.wait_for(event.wait(), timeout=timeout)
  File "C:\ProgramData\Anaconda3\lib\asyncio\tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
----------------------------------------
Traceback (most recent call last):
  File "server.py", line 367, in <module>
    serve()
  File "server.py", line 362, in serve
    server.serve_forever()
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 237, in serve_forever
    self._handle_request_noblock()
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "server.py", line 331, in finish_request
    self.handler.__init__(request, client_address, self)
  File "C:\ProgramData\Anaconda3\lib\socketserver.py", line 720, in __init__
    self.handle()
  File "C:\ProgramData\Anaconda3\lib\http\server.py", line 427, in handle
    self.handle_one_request()
  File "server.py", line 104, in handle_one_request
    super().handle_one_request()
  File "C:\ProgramData\Anaconda3\lib\http\server.py", line 415, in handle_one_request
    method()
  File "server.py", line 273, in do_POST
    self.handle_api()
  File "server.py", line 248, in handle_api
    self.printer.connect(name, address)
  File "D:\takeout\cat-printer-bare-0.6.3.0\cat-printer\printer.py", line 356, in connect
    self.loop(
  File "D:\takeout\cat-printer-bare-0.6.3.0\cat-printer\printer.py", line 327, in loop
    results.append(self._loop.run_until_complete(future))
  File "C:\ProgramData\Anaconda3\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
asyncio.exceptions.CancelledError
```

M1ghty-boy, 2023-08-01:
This issue occured both on the latest build and when running server.by after cloning the main branch

NaitLee, 2023-08-02:
This part of the stack trace:

```
File "server.py", line 248, in handle_api
    self.printer.connect(name, address)
```

tells that it timed out while connecting to the printer.

I can’t tell of your printer model is supported yet. Such thermal printers differs a lot, though all under similar colorful shell.

For Cat-Printer, the way telling if a device is supported is not well designed from today’s view: It should inspect the device’s parameters instead (BLE services, and/or characteristics), rather than looking at device name (typically 2 letters following 2 digits).
Sadly this project is not actively maintained at the moment.

The error is kinda like [this one](https://github.com/NaitLee/Cat-Printer/issues/28#issuecomment-1146546708), not sure if its relative, but just suspect the bluetooth adapter or operating system could also go wrong. (Maybe some Bluetooth adapters don’t support BLE well; Windows needs much more time for Bluetooth scanning & connection, compared to GNU/Linux or Android)

NaitLee, 2023-08-27:
You may visit this new Web app (with Chrome/Chromium/Edge etc.) and see if it will discover your printer: https://print.unseen-site.fun/

If yes, try to print with it; if not, then most likely your printer isn’t “cat printer” compatible.