NaitLee/Cat-Printer #76: how to use in windows?

Ern-Ani, 2023-08-12:
Is there any video teaching how to use it on windows?

NaitLee, 2023-08-12:
There isn’t yet, but I’m planning to make a video.

Before that reading #65 could help, too. Tell me if it really helps.

I’m also planning to make a new application, that can make it possible to print from just a web page, without installing anything (#52)

Ern-Ani, 2023-08-16:
Thanks for your reply to my question, unfortunately I'm too dumb to understand these walkthroughs, I guess I'll have to wait until your idea becomes more user friendly for the dumb user.

NaitLee, 2023-08-27:
I’ve made a Web app for Cat Printers: https://print.unseen-site.fun/

You just need a Chromium-based browser and visit it, e.g. Microsoft Edge or Google Chrome.

It will have more and more features in the future.