NaitLee/Cat-Printer #77: "Folder `printer_lib` is incomplete or missing, please check" error on MacBook Air M1

egor835, 2023-08-16:
[image]
What can I do with that? printer-lib is present, I use latest bare release, unpacked with default mac os archiver

NaitLee, 2023-08-16:
You may try to remove [this](https://github.com/NaitLee/Cat-Printer/blob/9a29e401fcf9c004c2b949ffd5fae4be679fcd95/printer.py#L90) try-except block and run `printer.py` again, which could tell more about what's really wrong.

i.e. change these:

```python

try:
    from printer_lib.models import Models, Model
    from printer_lib.commander import Commander, reverse_bits
    from printer_lib.text_print import TextCanvas
except ImportError:
    fatal(
        i18n('folder-printer_lib-is-incomplete-or-missing-please-check'),
        code=ExitCodes.IncompleteProgram
    )
```

to:

```python
from printer_lib.models import Models, Model
from printer_lib.commander import Commander, reverse_bits
from printer_lib.text_print import TextCanvas
```

NaitLee, 2023-08-17:
It’s found that 0.6.3.0 “bare” bundle is missing a critical file.

Sincere apologize for that 🤥 At the moment you can clone this repository directly and run stuffs inside.

NaitLee, 2023-08-17:
yes other os can’t run that bundle either