NaitLee/Cat-Printer #80: ESP32 support?

vr6syncro, 2023-08-21:
Hey is there a way to support ESP32? or can we add a ESP32 as a handle between Webinterface and Bluetooth? I wanna docker that webui and send it via wlan. The ESP32 have Bluetooth to communicate with the printer.

Any idea? I haven't delved deeper into the program code yet

vr6syncro, 2023-08-22:
Hey, the esp just need to be a middleman wifi to Bluetooth. So i think that should be easy because it is all existing we need only to Split the file that send data over bt and let the esp do it and add the middlemen software

NaitLee, 2023-08-22:
The current web frontend only produces image data, it’s the python backend produces bluetooth traffic.

The backend could dump its traffic. (without really connecting to printer, i.e. fake run)

Example command:
```bash
echo 'Hello, world!' | python printer.py -f GB02 -m -t 1,unicode,pf2 -
```
would produce a `traffic.dump`, as well as `dump.pbm` for the printed image.
(you need a [pf2 font bundle](https://github.com/NaitLee/Cat-Printer/files/11184775/pf2.zip) placed in cat-printer root dir for this command to run)

Though, in order to cleanly split the data, the dump format should change a bit, to include length of a “piece” in a leading byte. (it’s a whole binary chunk, while we should only send complete command “pieces” to printer)

Let me see if I can make some prototype ESP32 code in the following days :)

vr6syncro, 2023-08-23:
Thanks, i will Look into the Code aswell when i have more time

NaitLee, 2023-08-23:
Just now roughly modified [this piece of code](https://github.com/espressif/esp-idf/blob/32472536715d674c160a2565795895e0273f8cde/examples/bluetooth/bluedroid/ble/gatt_client/main/gattc_demo.c) to work with cat printers to print some fixed data.

Stuffs attached are code and compiled firmware to let an ESP32 scan & connect to a cat printer, print embedded “Hello, world!” bitmap & traffic data and stay there. Log is available via ttyUSB in baudrate 115200.

Flash with this command:
```
python3 esptool.py write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x1000 bootloader.bin 0x10000 firmware.bin 0x8000 partitions.bin
```

I’m personally using [PlatformIO](https://platformio.org/) and [IDF framework](https://idf.espressif.com/). The firmware is created using a preset config called “esp32thing” though I guess it’s generic.
(My chip is `ESP32-D0WD-V3 (revision v3.0)`)

I’m not professional and please point out anything wrong 🤣

[cat-printer-esp32-hello-world.tar.gz](https://github.com/NaitLee/Cat-Printer/files/12417664/cat-printer-esp32-hello-world.tar.gz)

NaitLee, 2023-08-28:
There’s already a library that seemingly supports cat printers on embeddable chips: https://github.com/bitbank2/Thermal_Printer

vr6syncro, 2023-08-30:
hey, i installed the Thermal_Printer libary but had to modify it to work with MX06 printer.  But i want a webinterface or connection to my smarthome.. the it seems i must program this by myself.