NaitLee/Cat-Printer #81: wich printer?

vr6syncro, 2023-08-26:
Hey, i need to buy a new printer and searching a version that can handle a bigger diameter of the printerpaper.
I think the Cat Printer can handle 35 diameter? Is there a bigger one for 40 or even 50?

Also i searching a Version with more space in the case. I googled GB01, GB02, GB03, GT01, YT01, MX05, MX06, MX08 but there are not much infos.

So @NaitLee  maybe you have more infos?

NaitLee, 2023-08-26:
Unfortunately, I have no much info about these printers.

All “Cat Printers” which I know have the same paper width, so if you get a “big” printer, even if it’s cat-printer compatible (unlikely), it won’t work with the current version of the software, because paper width is a fixed value in the program.

(Fixing it should be relatively easy via a few hack, though)

It is still unknown whether those “not-a-cat” printers by same vendors have the same protocol. True we can have a look by buying one, but in case investigation fails, it means either money waste or forced to use proprietary software.

Another option is to buy printers that are totally unrelative to cat printers — just make sure you are okay with their software.

vr6syncro, 2023-08-26:
I think we're confusing something. I mean the diameter not the width. dahher has nothing to do with the software because then the role is simply thicker
sorry if i wrote it confusingly

NaitLee, 2023-08-26:
Ohh, I went through too quickly 🙃

As for a “bigger roll”… As far as I know, cat printers share almost the same shell, so their inner size are similar.

The best I can imagine is to disassemble the printer shell and simply feed paper from the back 🤣 or 3D print a bigger shell if that’s an option.

vr6syncro, 2023-08-26:
mh ok i think i will do a 3d print thing... which is the best option that i can buy? or is working best?

NaitLee, 2023-08-26:
Indeed I’m unfamilar with 3D printing. I guess there’s no existing model to print, and you may need to make your own.

Probably a 3D printing store, if available, can make one.
I’m unclear about the process or price though.

vr6syncro, 2023-08-26:
hey i can do the 3d print stuff... maybe another misunderstood i mean wich thermo printer (cat printer) is the best working for thig project. sorry for that again.

NaitLee, 2023-08-27:
Ohh 🤣 The GB0x, MX0x etc. are very similar and “just works”.

However, likely that stores won’t tell you if they are “GB02” or “MX06”, but some codes like “C9” or so.

It’s (#69) said that [this](https://www.amazon.com/gp/product/B092Z4YQVN/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1) one just works.

vr6syncro, 2023-08-27:
just ordered one.. will reply if i get news

closed 2023-08-30 by vr6syncro