NaitLee/Cat-Printer #82: Suggestion: Paper feed adjustment

matrixes, 2023-09-05:
https://github.com/NaitLee/Cat-Printer/blob/9a29e401fcf9c004c2b949ffd5fae4be679fcd95/printer.py#L471

I have a GT01 printer that has the problem that if I set `self.feed_paper(128)` the bottom of the print is missing when tearing of the paper. If I set it to `self.feed_paper(256)` there is a lot of whitespace after the image which wastes paper.

Is there a way to control the feed length more precisely?

NaitLee, 2023-09-06:
The feeding was known to have problem on some models. At the moment you can add `GT01` to [here](https://github.com/NaitLee/Cat-Printer/blob/9a29e401fcf9c004c2b949ffd5fae4be679fcd95/printer_lib/models.py#L30) to workaround it.

As stated in #78, the code base will be refactored or rewrote in the future, prior to that I couldn’t add new features and release new versions.

And… I want to know if my new web app for cat printers works for you, especially the feeding behavior. Please test it: https://print.unseen-site.fun/

matrixes, 2023-09-06:
I did two tests

# Modifying models.py

This gave the same result as setting `self.feed_paper(256)`.

# Testing https://print.unseen-site.fun/

This gives the same result as the default; It does not feed enough, so that when tearing the paper the image is cut off at the bottom.

If it would be possible to have a setting for X length of extra feed, that would be preferable for my use. That way I can make adjustments for each print.

NaitLee, 2023-09-06:
I suspect I didn’t implement “feed” protocol in a right way. I didn’t had time to check it, though.

There’s [the same problem](https://github.com/NaitLee/Cat-Printer/issues/47#issuecomment-1427466206) for some other models.

It’s time for me to look back to other cat-printer projects that helped me. I’ll let you know if I made progress.

NaitLee, 2023-09-09:
I forced feed length data to be 2 bytes, as in other implementations, in the [web app](https://print.unseen-site.fun/). (it’s codebase is new and easier to change)

If it works then we found the solution.

matrixes, 2023-11-08:
I tested https://print.unseen-site.fun/ again and the feed works much better! Is there any way to run that code locally? As far as I can tell it works well enough for me.

NaitLee, 2023-11-10:
I “back ported” the possible fix from the rewrite code to main branch.

You may give it a test.