NaitLee/Cat-Printer #83: Suggestion for improvement.

erashmika, 2023-09-15:
@NaitLee 
Can you create a **DRIVER FOR WINDOWS** that can work well with these mini printers?

Then we can get any print easily without using any 3rd party software. Like a normal printer. (Plug And Play)
Also, we can connect the mini printer to a POS software and print something like a receipt very easily.

I hope you will give a favorable response to my proposal soon. Also, I hope that the driver I mentioned will be introduced in your next update. I appreciate your hard work for this. Thank you very much. ❤️❤️❤️

NaitLee, 2023-09-16:
tl;dr: A plug-and-play driver couldn’t be made in a short span of time. Workaround exists, but may not completely satisfy your need.

Sadly, after looking around, I have to tell these:

Windows development needs time — and much more for inexperienced ones. And, MS have deprecared the [“printer driver”](https://learn.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver) concept, in favor of a “print support app”.

The bluetooth printer is special — it doesn’t want traditional hardware “port” communication, but needs Bluetooth Low Energy. That’s something more to work on.

One workaround is, let your POS software print (save) its content to file (pdf or so), convert the file to image, and print it through this software (or [this web app](https://print.unseen-site.fun/)) manually.

Another workaround is: [add the printer in CUPS on a Unix machine](https://github.com/NaitLee/Cat-Printer/discussions/8#discussioncomment-2557843), and configure CUPS to accept network printing requests (complex to do), then add CUPS server as a network printer on Windows.

Though, IPP itself is also complex, it isn’t guaranteed to work properly.

I personally really don’t have time to make a driver or complete built-in IPP protocol, to satisfy your need “true driver for windows.”

And, please, I beg you, don’t hurt me. I revived this project 1 year ago just because someone’s request had bailed me out from previous one.

In my mind in case serious workload is needed, you can always contact for business solutions.

verypurple, 2023-11-02:
@NaitLee I am curious if you tried this driver with a kitty printer? https://www.netum.net/pages/thermal-receipt-printer-setup-guide

I have one of their devices and it is just serial over USB. In a kitty printer teardown I saw on YouTube they said USB data lines are connected to the main IC so maybe it works?

NaitLee, 2023-11-02:
There are two kinds of “cat printers” known.

One kind:
- model named `AAXX`, where `A` for an alphabet and `X` for a digit
- *no serial connection or so*

Another kind is:
- model named `YHK-*`
- *known to [act as USB device](https://github.com/JJJollyjim/catprinter#yhk-c200)*

For Bluetooth, my solutions work for the former one, and [this web app](https://print.unseen-site.fun/) will tell you whether it works or not.
And the later one needs entirely different solution, like [this one](https://github.com/abhigkar/YHK-Cat-Thermal-Printer). Unfortunately seems that yours is this one and I probably couldn’t help out.

Sadly I couldn’t help you to test that driver either, because I don’t have `YHK-*` in hand. Probably you may just ask that Youtuber… :)

verypurple, 2023-11-03:
Thank you so much for offering help but my printer works! I just like the kitty printers and was curious to learn more about them. I am leaving the video I mentioned https://www.youtube.com/watch?v=4dRGZEcAp38 the printer is labeled GW-J550 and I don't see on the list. A new type maybe?