NaitLee/Cat-Printer #87: Doesn't work on MacOS Ventura intel

Tintouin, 2023-11-03:
It shows this error report when ran either with or without sudo. 

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Python [3911]
Path:                  /Library/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.10.4 (3.10.4)
Code Type:             X86-64 (Native)
Parent Process:        Exited process [3910]
Responsible:           iTerm2 [3589]
User ID:               0

Date/Time:             2023-11-03 22:47:52.1922 +0100
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Bridge OS Version:     8.0 (21P365)
Anonymous UUID:        D93E1030-A677-99C5-1234-80C4E07D1BC9

Sleep/Wake UUID:       08C31E28-E7DC-46AC-B570-6CFB815C071C

Time Awake Since Boot: 12000 seconds
Time Since Wake:       2973 seconds

System Integrity Protection: enabled

Crashed Thread:        2  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace TCC, Code 0 
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff80da6f0de __psynch_cvwait + 10
1   libsystem_pthread.dylib       	    0x7ff80daab758 _pthread_cond_wait + 1242
2   Python                        	       0x10527ba55 PyThread_acquire_lock_timed + 293
3   Python                        	       0x1052f686a acquire_timed + 234
4   Python                        	       0x1052f6aac lock_PyThread_acquire_lock + 44
5   Python                        	       0x1050d005a method_vectorcall_VARARGS_KEYWORDS + 186
6   Python                        	       0x10520201f call_function + 175
7   Python                        	       0x1051fb22a _PyEval_EvalFrameDefault + 34522
8   Python                        	       0x1051f11df _PyEval_Vector + 383
9   Python                        	       0x10520201f call_function + 175
10  Python                        	       0x1051fb22a _PyEval_EvalFrameDefault + 34522
11  Python                        	       0x1051f11df _PyEval_Vector + 383
12  Python                        	       0x10520201f call_function + 175
13  Python                        	       0x1051fb22a _PyEval_EvalFrameDefault + 34522
14  Python                        	       0x1051f11df _PyEval_Vector + 383
15  _objc.cpython-310-darwin.so   	       0x105d24872 PyObject_Vectorcall + 178
16  _objc.cpython-310-darwin.so   	       0x105d2365b pysel_vectorcall + 363
17  Python                        	       0x10520201f call_function + 175
18  Python                        	       0x1051fb2c4 _PyEval_EvalFrameDefault + 34676
19  Python                        	       0x1051f11df _PyEval_Vector + 383
20  Python                        	       0x1050c1000 _PyObject_FastCallDictTstate + 96
21  Python                        	       0x105155dd9 slot_tp_init + 201
22  Python                        	       0x10514c662 type_call + 322
23  Python                        	       0x1050c1acd _PyObject_Call + 109
24  Python                        	       0x1051fb4f2 _PyEval_EvalFrameDefault + 35234
25  Python                        	       0x1050dd93b gen_send_ex2 + 203
26  _asyncio.cpython-310-darwin.so	       0x104fa693b task_step_impl + 427
27  _asyncio.cpython-310-darwin.so	       0x104fa6720 task_step + 48
28  Python                        	       0x1050c0d74 _PyObject_MakeTpCall + 132
29  Python                        	       0x105222bc1 context_run + 81
30  Python                        	       0x1051284cb cfunction_vectorcall_FASTCALL_KEYWORDS + 75
31  Python                        	       0x1051fb5b7 _PyEval_EvalFrameDefault + 35431
[... 678 lines cut ...]

Model: MacBookPro16,1, BootROM 2020.0.1.0.0 (iBridge: 21.16.365.0.0,0), 6 processors, 6-Core Intel Core i7, 2,6 GHz, 16 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Display: Color LCD, 3072 x 1920 Retina, Main, MirrorOff, Online
Graphics: AMD Radeon Pro 5300M, AMD Radeon Pro 5300M, PCIe, 4 GB
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E1
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E1
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Dec  9 2022 17:02:25 version 9.30.492.0.32.5.87 FWID 01-e7856862
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: 2.4G Keyboard Mouse
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

```

NaitLee, 2023-11-04:
As said by this line:

```
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.
```

It’s said that you need to give your *terminal app* bluetooth permission: https://github.com/hbldh/bleak/issues/761#issuecomment-1036437107
And there’s screenshot for doing that: https://github.com/orgs/pybricks/discussions/583#discussioncomment-1933666

---

And another promotion to kitty-printer: https://print.unseen-site.fun/
It probably works better in most scenarios.

Tintouin, 2023-11-04:
Thanks it worked but I had to also enable bluetooth on Opera, though the Preference app has changed and the layout is different you may want me to provide screenshots, also you should mention it in the macOS part of the readme.md.

closed 2023-11-04 by Tintouin