NaitLee/Cat-Printer #88: MX06 can't discover device

pentirecords, 2023-11-06:
Not sure if im doing something wrong but i have been trying to connect this MX06 printer with the latest main Cat-Printer on localhost which seems to be working normally besides the fact that i cant find my MX06 no matter how much i scan..

And also tried the https://print.unseen-site.fun/ (kitty printer) but that also can't find my MX06...

I use a Bluetooth adapter for my windows 11 pc

If anyone has any suggestions as to why i cant see the mx06 anywhere that would be awesome since i really like this app but unfortunately i cant even connect to my printer lmao, thanks for making it anyway

NaitLee, 2023-11-06:
You may try kitty-printer on other devices (like your mobile phone or your friends’ PC), see if they can discover your MX06.

Adapter, OS, and/or PC all can be problematic, we just need to find the problem.

strickinato, 2023-12-20:
@pentirecords - any luck?

I also have an MX06 that is not connecting.

Urufusan, 2024-01-04:
Mind testing on a Linux distro (like Ubuntu)?

pentirecords, 2024-01-04:
For me personally the problem was on my end since i just got a laptop yesterday and tried again on that one with just the kittyprinter web version and that one worked perfectly!
I assume its just that my desktops bluetooth adapter might not be BLE compatible or something

closed 2024-01-04 by pentirecords