NaitLee/Cat-Printer #91: Pillow for image conversion?

Urufusan, 2024-01-02:
IMO, using ImageMagick is a bit of a hassle (e.g. setting up the permissions, having to install it on a fresh system (or just can't, because of an environment like SteamOS)).

Why can Pillow be used for image conversion to pbm? I managed to create a working implementation in my fork. Could it be that Pillow was avoided because of the contribution rule that states that no big dependencies should be used?

NaitLee, 2024-01-03:
In the past Cat-Printer is designed to be portable so that one don't need to install anything from pip to use it.

ImageMagick is pre-installed on many systems, and installing it is simple in *most* cases.

Though, you are right, external deps are not everywhere. I thought to rewrite Cat-Printer so that it can work with pip, and further more *optional* deps like PIL will be acceptable. But the rewrite isn't going on for months, I'm not active coding these days.

I'd seek time to complete it, if I can. When that happens, we can implement things with optional PIL etc.

Urufusan, 2024-01-03:
Perhaps we can use PIL in the current codebase as a fallback? Basically, the logic that checks for ImageMagick could fall back to using a tiny hack [like this one](https://github.com/Urufusan/Cat-Printer/blob/main/new_pbm_generator.py), this is how I made a [working library example in my fork](https://github.com/Urufusan/Cat-Printer/blob/main/test_library.py)

Urufusan, 2024-01-03:
I'd be happy to make a nicer patch and put it in a PR.

NaitLee, 2024-01-03:
Yes, you may do it :)

Probably I should update the contribution rule a bit... Any dependencies are ok as long as they are optional, and that's easy to do.