NaitLee/Cat-Printer #93: MX10 Cat Printer Support?

firepowerdev, 2024-02-01:
Hello, I bought a new cat thermal printer last month and its model is MX10, but when I tried to use it in the program you made, I realized that it was not supported. I wonder if there will be an update to your program for this model?

NaitLee, 2024-02-01:
You may try to add `MX10` to [this line of code](https://github.com/NaitLee/Cat-Printer/blob/a7dee14c96c26b6c241b9d7000dde5325e3e10aa/printer_lib/models.py#L22), and see if your printer can be used directly after that.

Or try to use [this web app](https://print.unseen-site.fun/), see if your printer can be found by your browser. If yes, then mostly it will just work.

firepowerdev, 2024-02-01:
Thanks, adding MX10 into the code solved my problem.

closed 2024-02-01 by firepowerdev