NaitLee/Cat-Printer #95: Way to interact with the android app via python?

diggedypomme, 2024-02-14:
The android version of this works great via the android phone app, but via termux and running the python, the bluetooth doesn't work (due to termux, not this repo). I am running a separate python app and want to print out - is there a setup that can be used to interact and push images to the backend of the android app without using the web front end? I was trying to get it working via post to /set but I'm struggling with it, so any advice would be great. Thank you

NaitLee, 2024-02-14:
The android app do have the server listening on `127.0.0.1:8095`, so leaving the app background and doing http requests with programs will work.

You can see API endpoints [in the server code](https://github.com/NaitLee/Cat-Printer/blob/2ff61f7b8d9bde594abbc98548addc8e6ca67fcc/server.py#L212C9-L212C19), and see how to interact with it by seeking for calls to [this function in frontend code](https://github.com/NaitLee/Cat-Printer/blob/2ff61f7b8d9bde594abbc98548addc8e6ca67fcc/www/main.js#L186).

Any language would be ok for this task. If you need any example code, just ask :)

diggedypomme, 2024-02-14:
Awesome thank you. I got it working - printed my first hello message.

bgcheck, 2024-04-12:
Quick and dirty example
https://gist.github.com/bgcheck/09d093633ddd2349f160153d12c1bd54