NaitLee/Cat-Printer #96: Index out of range exception upon printing

alessandro-antonelli, 2024-02-19:
I'm encountering an uncaught exception that prevents me from printing on the Windows "bare bone" version.

Error message displayed on app:

```
Error: API Failure
    at Main.handleBluetoothProblem (http://127.0.0.1:8095/~every.js:1484:17)
{
    "name": "Exception",
    "details": "bytearray index out of range"
}
```

Error message displayed on terminal:

```
Serving at http://127.0.0.1:8095/
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 54735)
Traceback (most recent call last):
  File "C:\Users\aless\AppData\Local\Programs\Python\Python312\Lib\socketserver.py", line 318, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Users\aless\AppData\Local\Programs\Python\Python312\Lib\socketserver.py", line 349, in process_request
    self.finish_request(request, client_address)
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\server.py", line 329, in finish_request
    self.handler.__init__(request, client_address, self)
  File "C:\Users\aless\AppData\Local\Programs\Python\Python312\Lib\socketserver.py", line 761, in __init__
    self.handle()
  File "C:\Users\aless\AppData\Local\Programs\Python\Python312\Lib\http\server.py", line 436, in handle
    self.handle_one_request()
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\server.py", line 104, in handle_one_request
    super().handle_one_request()
  File "C:\Users\aless\AppData\Local\Programs\Python\Python312\Lib\http\server.py", line 424, in handle_one_request
    method()
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\server.py", line 271, in do_POST
    self.handle_api()
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\server.py", line 219, in handle_api
    self.printer.print(io.BytesIO(body))
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\printer.py", line 414, in print
    self._print_bitmap(printer_data)
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\printer.py", line 478, in _print_bitmap
    self._prepare()
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\printer.py", line 458, in _prepare
    self.set_energy(self.energy)
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\printer_lib\commander.py", line 142, in set_energy
    self.send( self.make_command(0xaf, int_to_bytes(amount, length=2)) )
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\aless\AppData\Roaming\Cat-Printer\printer_lib\commander.py", line 52, in int_to_bytes
    b[p] = i & 0xff
    ~^^^
IndexError: bytearray index out of range
----------------------------------------
```

Setup where error occurs:

- Printer: MX06 (exact article: https://it.aliexpress.com/item/1005005275882875.html )
- PC: tried both in a Windows 10 desktop fitted with a Bluetooth 5.3 USB adapter, and in a Windows 11 laptop with built-in Bluetooth 5.2
- Error does **not** occur in the Android app (app version 0.6.3.0 / Android 9 / Samsung Galaxy S8)

Steps to reproduce:

1. Install Python from https://www.python.org/ftp/python/3.12.2/python-3.12.2-amd64.exe
2. Install Bleak with `py -m pip install bleak` on the command prompt
3. Go in `C:\Users\aless\AppData\Roaming` and execute `git clone https://github.com/NaitLee/Cat-Printer.git`
4. Execute in command prompt `python server.py`
5. Visit http://127.0.0.1:8095/ in MS Edge browser
6. Click "Show more options" -> "Scan"
7. Select `MX06-10D6` from the dropdown
8. Insert anything (either pic or text) and click Print
9. The error message is shown and nothing is printed

NaitLee, 2024-02-21:
I can’t personally reproduce it locally, but I just added a few checks to that errored function, which would probably give more helpful info.

Please try:
- run `git pull` in the repo and try to reproduce it, see if there’s any error
- after that, in the web ui, try setting the “Strength” slider to any value (in “more options”), then try to print again

You may also need to provide which browser you are using, if the error persists.

alessandro-antonelli, 2024-03-04:
It seems that commit e976b0d fixed it! I tried the new version, and I could print without any error message. The print is successfull even without touching the Strength slider, at least on my system. Thank you a lot!

NaitLee, 2024-03-04:
Glad to hear that :)

No idea what caused that though.

doragasu, 2024-03-16:
I still have this error with latest commit, or at least something very similar:

```
$ python server.py 
Serving at http://127.0.0.1:8095/
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 48192)
Traceback (most recent call last):
  File "/usr/lib/python3.11/socketserver.py", line 317, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.11/socketserver.py", line 348, in process_request
    self.finish_request(request, client_address)
  File "/tmp/Cat-Printer/server.py", line 329, in finish_request
    self.handler.__init__(request, client_address, self)
  File "/usr/lib/python3.11/socketserver.py", line 755, in __init__
    self.handle()
  File "/usr/lib/python3.11/http/server.py", line 436, in handle
    self.handle_one_request()
  File "/tmp/Cat-Printer/server.py", line 104, in handle_one_request
    super().handle_one_request()
  File "/usr/lib/python3.11/http/server.py", line 424, in handle_one_request
    method()
  File "/tmp/Cat-Printer/server.py", line 271, in do_POST
    self.handle_api()
  File "/tmp/Cat-Printer/server.py", line 219, in handle_api
    self.printer.print(io.BytesIO(body))
  File "/tmp/Cat-Printer/printer.py", line 414, in print
    self._print_bitmap(printer_data)
  File "/tmp/Cat-Printer/printer.py", line 478, in _print_bitmap
    self._prepare()
  File "/tmp/Cat-Printer/printer.py", line 458, in _prepare
    self.set_energy(self.energy)
  File "/tmp/Cat-Printer/printer_lib/commander.py", line 149, in set_energy
    self.send( self.make_command(0xaf, int_to_bytes(amount, length=2)) )
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/Cat-Printer/printer_lib/commander.py", line 55, in int_to_bytes
    raise f'int_to_bytes: {i} > {max_value}'
TypeError: exceptions must derive from BaseException
----------------------------------------
```
Maybe something in python internals has changed because I was printing about 2 or 3 weeks ago without a problem. I am using up to date Archlinux.