NaitLee/Cat-Printer #109: Add support for X6 mobile printers

proffalken, 2024-07-08:
I recently purchased an [X6 Bluetooth Printer](https://www.aliexpress.com/item/1005006569536699.html) from AliExpress.

If I run the server and set `Test Unknown Device`, it works from both the UI and from CUPS (I followed [this comment](https://github.com/NaitLee/Cat-Printer/discussions/8#discussioncomment-2557843) to get it working as a generic IPP printer.)

I'd now like to integrate it into my network via a Pi running Cups and Cat-Printer - how do I go about working out the things you need to know in order to get it added to the list of supported devices?

proffalken, 2024-10-09:
@NaitLee - apologies for the ping, just wondering if you've got a list on what to do here?

izissise, 2025-06-18:
I couldn't make it work through unknown device, the UI says printing and then it error out

SilverGreen93, 2025-12-25:
I am too interested in support for X6h on all OSes

Dejniel, 2026-01-09:
You can check my TiMini-Print project — it already supports the X6 by talking directly over its Bluetooth protocol (not IPP): [https://github.com/Dejniel/TiMini-Print](https://github.com/Dejniel/TiMini-Print)